bdring/Grbl_Esp32 #410: Squaring_Mode_Dual

reynolds087, 2020-05-20:
Hi,
I am trying to find documentation for the dual squaring mode to home the two sides of a gantry with each side having an independent limit switch. I also am not sure where to find documentation about the pin assignments of an esp32 dev module and their corresponding pins in the GRBL firmware + how to customize and the implications of doing so. I would think all of this information is somewhat related and necessary in order to implement the dual squaring feature, but I can't find any of it. Could you please help?

bdring, 2020-05-20:
The best resource we have right now are examples. 

The MPCNC uses it for X and Y axes. Both switches on an axis connect to the same I/O pin. It uses a separate step pin for each side. It uses a common direction pin for each side.
https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/Machines/mpcnc_v1p2.h
https://github.com/bdring/Grbl_ESP32_MPCNC_Controller/blob/master/source/Schematic_1p2p1.pdf

This one is similar but uses a separate direction pin for each side
https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/Machines/spi_daisy_4axis_xyyz.h
https://github.com/bdring/4_Axis_SPI_CNC/blob/master/docs/1p3/SPI_4Axis_V1p3_Schm.pdf

I'll try to get a wiki page written for this soon.

Are you designing new hardware or using an existing controller?

reynolds087, 2020-05-20:
I have this board: 

https://www.amazon.com/gp/product/B07QCP2451/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I don't know how to begin wiring it up. Is there a diagram of the default pins? I have external five TB6600 drivers (two Y axis drivers that I want to operate independently for squaring, and two X axis that I want to run in parallel/inverted on the same lead screw). 

I am not sure how to specify the example machine.
Do I just change this line in machine.h and set it to the example file I want to use?

#include "Machines/test_drive.h"

bdring, 2020-05-20:
Yes, you change that line in machine.h to an existing machine definition file or create your own. All of the machine files are in the Machines folder.

Grbl_ESP32 will support the Ganged and Squared mode for the Y and the Ganged only mode for the X. The direction is tricky, but easily fixed in the wiring of the motors to the drivers (swap one coils wires)

The TB6600 drivers are odd balls. They are unsupported and every manufacturer sets them up differently. People have used them, but require a lot of our support. I cannot help you with those.

I am working on the wiki page, so stay tuned...

reynolds087, 2020-05-20:
Thanks, ok please let me know when the wiki page goes up. 

So it's not as simple as just wiring to the inputs of the TB6600 and provide the external power? I thought it was kind of like when you take an audio signal from your phone and plug it into an active speaker and it gets amplified.

bdring, 2020-05-20:
https://github.com/bdring/Grbl_Esp32/wiki/Motor-Ganging-and-Axis-Squaring

ithinkido, 2020-05-21:
> The TB6600 drivers are odd balls. They are unsupported and every manufacturer sets them up differently. People have used them, but require a lot of our support. I cannot help you with those.
> 

https://www.makerguides.com/tb6600-stepper-motor-driver-arduino-tutorial/

About half way down this article is an interesting piece about "fake" TB6600 drivers that are actually a different chip. I can see this being a big issue when setting things up and then realising that it will not work. It might be handy to add a line about this in the  wiki so that people can get one more idea where to check when trouble shooting .

bdring, 2020-05-21:
@ithinkido   If you want to write a blurb for the [Wiki page]9https://github.com/bdring/Grbl_Esp32/wiki/Stepper-Motor-Drivers), I can include it.

I know there are fake chips. The TB6600 chip is not bad, but the way they integrate them into the package is a mess. They are looking for the the lowest cost driver that actually looks like a "real" driver.

They are unsupported because the person you bought them from is unlikely to ever help you. People should not expect open source developers to fill that role.

ithinkido, 2020-05-21:
Sure, 
[Stepper-Motor-Drivers.zip](https://github.com/bdring/Grbl_Esp32/files/4663621/Stepper-Motor-Drivers.zip)

jaysettle, 2020-05-28:
@reynolds087, I can help you out on your setup if needed. I have a similar configuration.

reynolds087, 2020-05-30:
@jaysettle, thanks. I'll give it a try and report back if needed.

closed 2020-11-02 by bdring