bdring/Grbl_Esp32 #411: Parameters 100-102 Not Showing Correct Values After Firmware Flash

reynolds087, 2020-05-25:
Please answer the following questions.

What version of the firmware are you using?

I don't know how to check, but it's the master branch from 2 days ago.

Is the problem repeatable?

I think so.

Under what conditions does the bug occur?

Change the steps/mm in your machine file, and then update machine.h to use the machine file. Flash the firmware and check parameters with $+ and it shows 100mm for $100-$102

```C++
  #define EASIER_TO_READ true
```

reynolds087, 2020-05-25:
I have these lines in my example file:

`#define DEFAULT_X_STEPS_PER_MM 400.0
#define DEFAULT_Y_STEPS_PER_MM 400.0
#define DEFAULT_Z_STEPS_PER_MM 400.0
`
But after flashing the firmware it still shows 100 for $100, $101, and $102

MitchBradley, 2020-05-25:
Simple changing the defaults in the code and recompiling does not necessary change the values.  The compiled-in defaults are only used if the checksum on the EEPROM settings area is invalid.  Reloading the firmware does not necessarily invalidate the EEPROM.  You can force it to be invalidated, and reset to the compiled defaults, by executing $RST

reynolds087, 2020-05-25:
Thanks; ok, I will keep that in mind going forward. Feel free to close the ticket.

closed 2020-05-25 by MitchBradley

reynolds087, 2020-05-25:
Sorry one other thing, when I run `$RST` into the terminal I get an error:3
Is there any syntax I am missing

MitchBradley, 2020-05-25:
Sorry I forgot to say "$RST=$"

MitchBradley, 2020-05-25:
If you want to RTFM, here it is https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands

reynolds087, 2020-05-25:
Thanks, much appreciated!

reynolds087, 2020-05-25:
Something else I can't find and I've been searching the wiki including this page: https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

I can't find the stepper enable pins. Isn't that how I set up the holding current? For instance I set $1 to 255 so that the stepper will have current at all times, but nothing happens because I don't have a wire from my driver to the step enable pin on the ESP32. Is there a default pin, and is there a `#define` command I can use to change my machine file?

I'm sorry if this information is already available somewhere. I really did my best to search for it before asking.

MitchBradley, 2020-05-25:
Search for STEPPERS_DISABLE_PIN in the files in the Machines/ 
subdirectory. Most of the boards use GPIO 13 as a shared disable pin.  
In the past, individual enables have not be used due to pin limitations, 
but that will change with the new 6-axis I2S board, which has enough 
output pins to allow each driver to have its own enable.

That said, stepper enable is not necessary for initial current setting 
in most cases.  In the usual setups, the motor current is set with one 
of the following methods:

  * A small potentiometer on the driver module
  * A set of jumpers next to the driver module
  * Serial commands sent to a "smart" driver like a Triaminic