bdring/Grbl_Esp32 #412: Documentation Correction: NO/NC series+parallel switches

carneeki, 2020-05-26:
In the wiki entry for setting up IO pins under [Motor Ganging and Axis Squaring](https://github.com/bdring/Grbl_Esp32/wiki/Motor-Ganging-and-Axis-Squaring#io-pins) - 

The textual information is correct but the labels in the graphics are wrong. The graphics show NO in parallel and are labeled NO in series, while the NC switches are in series and labeled as parallel.

Additionally, I propose moving the contents of that section into the section on dual switches in [Setting Up Limit Homing Switches](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-Limit-Homing-Switch#dual-switches) and simply noting the requirement for dual switches and linking to that section.

Really great documenation though! I've been looking for something to replace my Pi+Grbl Hat and this is looking like a pretty good candidate :)

bdring, 2020-05-26:
I don't understand the problem. Please mark it up for reference.

[image]

MitchBradley, 2020-05-26:
In the picture, the words "series" and "parallel" should be swapped.  The picture on the left shows a parallel connection but is labeled "series", and vice versa

bdring, 2020-05-26:
Ahhh.

[image]

closed 2020-05-26 by bdring