bdring/Grbl_Esp32 #413: Suggest to consider the new ESP32-S2 module

ningzhang-github, 2020-05-26:
Hello guys,

Greatings from China! 
At work I am an vehicle SW test automation engineer, in my spare time I am a fan of CNC, GRBL, and ESP32 MCU applications such as Onstep(a telescope controller). When I were wondering around github I found this awsome project, which combined all the things I like to have. Outstanding job @bdring !

Back to the topic, Espressif the maker of ESP 8266/32 MCUs has released the new ESP32-S2 serial to market, it has 43 available GPIOs, more than the 34 pins(correct me if im wrong) of current ESP32. This should be greatly helpful for our Grbl_ESP32 project. Its small devkit is already available in the China domestic aliexpress store(Taobao), priced in ~7USD/pcs, should be also available in international Aliexpress.

Brief intro of ESP32-S2: https://www.espressif.com/en/news/espressif-announces-%E2%80%A8esp32-s2-secure-wi-fi-mcu

Info of its devkit: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html
Picture:
[image]

Just my two cnets, hope it helps.
Thanks again for your great effort to make this project happen.
 
Brs,
Ning

bdring, 2020-05-26:
It only has one core. I hear a dual core version is coming soon.

It would take a lot of work to refactor for one core and performance may not be as good.

MitchBradley, 2020-05-26:
I bought an S2 but made the mistake of getting the bare chip instead of a module. Originally I was excited by the fact that it has more I/Os, but it turns out that the number of extra I/Os is not enough to truly "change the game".  The recent I2S work is that "game changer"; it provides 32 extra outputs with fast, stable stepping, and could provide a similar number of extra inputs if needed.  The shift register chips that it uses are inexpensive.  They can be distributed around the board in ways that make PCB layout easier.

ningzhang-github, 2020-05-28:
Okay now I understand, I would like to close this thread since developers are all aware of the info I mentioned.
PS: Just read through the I2S I/O extension PR, brilliant! Can't wait to see its release.

closed 2020-05-28 by ningzhang-github