bdring/Grbl_Esp32 #415: Problems Compiling Firmware with new esp32s2 github branch

wdfortee, 2020-05-30:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)? Yes

What version of the Arduino IDE are you using?  Arduino IDE 1.8.12

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 

https://github.com/espressif/arduino-esp32/tree/esp32s2

Are you using the master branch of Grbl_ESP32? No

Have you made any edits or configuration changes (list them) to the  firmware? No

Please paste the compiler error text here:

sketch\serial.cpp: In function 'void serialCheckTask(void*)':
serial.cpp:136:17: error: 'vTaskEnterCritical' was not declared in this scope
                 vTaskEnterCritical(&myMutex);
                 ^~~~~~~~~~~~~~~~~~
sketch\serial.cpp:136:17: note: suggested alternative: 'vPortEnterCritical'
                 vTaskEnterCritical(&myMutex);
                 ^~~~~~~~~~~~~~~~~~
                 vPortEnterCritical
serial.cpp:138:17: error: 'vTaskExitCritical' was not declared in this scope
                 vTaskExitCritical(&myMutex);
                 ^~~~~~~~~~~~~~~~~
sketch\serial.cpp:138:17: note: suggested alternative: 'vPortExitCritical'
                 vTaskExitCritical(&myMutex);
                 ^~~~~~~~~~~~~~~~~
                 vPortExitCritical
sketch\serial.cpp: In function 'uint8_t serial_read(uint8_t)':
serial.cpp:170:5: error: 'vTaskEnterCritical' was not declared in this scope
     vTaskEnterCritical(&myMutex);
     ^~~~~~~~~~~~~~~~~~
sketch\serial.cpp:170:5: note: suggested alternative: 'vPortEnterCritical'
     vTaskEnterCritical(&myMutex);
     ^~~~~~~~~~~~~~~~~~
     vPortEnterCritical
serial.cpp:173:9: error: 'vTaskExitCritical' was not declared in this scope
         vTaskExitCritical(&myMutex);
         ^~~~~~~~~~~~~~~~~
sketch\serial.cpp:173:9: note: suggested alternative: 'vPortExitCritical'
         vTaskExitCritical(&myMutex);
         ^~~~~~~~~~~~~~~~~
         vPortExitCritical
serial.cpp:177:9: error: 'vTaskExitCritical' was not declared in this scope
         vTaskExitCritical(&myMutex);
         ^~~~~~~~~~~~~~~~~
sketch\serial.cpp:177:9: note: suggested alternative: 'vPortExitCritical'
         vTaskExitCritical(&myMutex);
         ^~~~~~~~~~~~~~~~~
         vPortExitCritical
In file included from sketch\grbl.h:39,
                 from sketch\stepper.cpp:26:
sketch\stepper.cpp: In function 'void stepper_init()':
nuts_bolts.h:27:14: error: invalid conversion from 'int' to 'timer_autoreload_t' [-fpermissive]
 #define true 1
              ^
sketch\stepper.cpp:503:26: note: in expansion of macro 'true'
     config.auto_reload = true;
                          ^~~~

bdring, 2020-05-30:
We do not support the single core esp32s

wdfortee, 2020-05-30:
I really don't need help with this.  I've reverted back to master branch of arduino-esp32. I just thought this information might be helpful for future when Espressif merges the esp32s2 branch.

With my limited knowledge on programming, I think I've managed to find out that "vTackEnterCritical" and "vTaskExitCritical" have been deprecated. Also, the "timer_config_t" struct's autoreload has been changed somehow, but I wasn't really able to dive into it more.

bdring, 2020-05-31:
Grbl_ESP32 puts tasks on both cores. You cannot compile it for 1 core.

closed 2020-05-31 by bdring