bdring/Grbl_Esp32 #416: Serial COMM only in Testdrive

wayneosdias, 2020-05-31:
Im having issue w getting serial comm to work w the machine defined anything other than test_drive.h. When test driving tx rx work using the general GRBL commands. Using Machines/3axis_v4.h or any other variant with in the machines folder and ESP32 only txs board properties on hard reset. The module doesnt seem to rx commands. Im not experienced w the Arduino IDE and dont care for it much. 

Can someone help me get rolling. Im after a 4 axis setup  with a pair of low freq gpio for controlling relays, switch frequency is not important. Can someone share a working machine file?

bdring, 2020-05-31:
We have no information to help us debug this problem. Please paste your startup text in a reply.
https://github.com/bdring/Grbl_Esp32/wiki/FAQ#submitting-bug-information

What hardware have you connected the ESP32 to? The machine definition file you are using requires specific hardware attached or floating switch inputs could be locking up the ESP32.

wayneosdias, 2020-05-31:
Hi Bart,  
Sorry, Im using a original Espressif board the "ESP32_Core_board_V2" and have duplicated the issue on a ESP-WROVER-KIT V3 board. Startup text appended at bottom of reply

Ive made some progress and have full TX RX working on a modified 3axis_v4.h file. I started by commenting out everything, recompiling and sequentially uncommenting each line. Ive gotten all the step/dir gpio uncommented w full TX RX. Comms are working but I havnt checked gpio hardware just yet. The end application is a CNC router with Clearpath servos.

Ill push on to the uncommenting ifdef's but first I want to try and figure out how use the Arduino IDE. Now Im using Visual Studio to modify Arduino file and unable to step thru, its taking too long and is clumsy to get thru...

Hardreset boot text;
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.2a Date 20200514]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Warning: Spindle output pin not defined]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.2a ['$' for help]

bdring, 2020-05-31:
You did not answer my question, but it does not appear that you do not have the ESP32 connected to a CNC controller board. 

It appears you have shown my the startup text after you made changes that made it work better. That does not help me.

Do not use pins 34-39 for anything unless you have resistors attached. https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins#input-only-no-pulluppulldown

wayneosdias, 2020-05-31:
Hey Bart, no hardware, cnc controller breakout or otherwise is connected to the ESP32, Its just a ESP32_Core_board_V2 connected to PC via a USB cable.

Heres boot text for RX not working ("#include "Machines/3axis_v4.h", no response to $$):
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.2a Date 20200514]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:PWM spindle Output:2, Enbl:22, Dir:-1, Freq:5000Hz, Res:13bits]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.2a ['$' for help]
[MSG:Check door]

Heres boot text for TX RX working (ifdefs commented out and the response to $$):
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.2a Date 20200514]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Warning: Spindle output pin not defined]

[MSG:Local access point GRBL_ESP started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.2a ['$' for help]
$0=3
$1=250
$2=0
$3=0
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=3
$24=200.000
$25=2000.000
$26=250
$27=1.000
$30=1000.000
$31=0.000
$32=0
$100=100.000
$101=100.000
$102=100.000
$110=1000.000
$111=1000.000
$112=1000.000
$120=200.000
$121=200.000
$122=200.000
$130=300.000
$131=300.000
$132=300.000
ok
ok

Heres the actual machine.h file that works:
`/*
    3axis_v4.h
    Part of Grbl_ESP32

    Pin assignments for the ESP32 Development Controller, v4.1 and later.
    https://github.com/bdring/Grbl_ESP32_Development_Controller
    https://www.tindie.com/products/33366583/grbl_esp32-cnc-development-board-v35/

    2018    - Bart Dring
    2020    - Mitch Bradley

    Grbl_ESP32 is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    Grbl is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with Grbl_ESP32.  If not, see <http://www.gnu.org/licenses/>.
*/

#define MACHINE_NAME            "ESP32_V4"

#define X_STEP_PIN              GPIO_NUM_12
#define X_DIRECTION_PIN         GPIO_NUM_14
#define Y_STEP_PIN              GPIO_NUM_26
#define Y_DIRECTION_PIN         GPIO_NUM_15
#define Z_STEP_PIN              GPIO_NUM_27
#define Z_DIRECTION_PIN         GPIO_NUM_33

#define LIMIT_MASK              B111
#define X_LIMIT_PIN             GPIO_NUM_17
#define Y_LIMIT_PIN             GPIO_NUM_4
#define Z_LIMIT_PIN             GPIO_NUM_16
/*
#ifdef HOMING_CYCLE_0
    #undef HOMING_CYCLE_0
#endif
#define HOMING_CYCLE_0 (1<<Z_AXIS) // Z first

#ifdef HOMING_CYCLE_1
    #undef HOMING_CYCLE_1
#endif
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))

#ifdef HOMING_CYCLE_2
    #undef HOMING_CYCLE_2
#endif

// OK to comment out to use pin for other features
#define STEPPERS_DISABLE_PIN    GPIO_NUM_13

#define SPINDLE_TYPE            SPINDLE_TYPE_PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_2   // labeled SpinPWM
#define SPINDLE_ENABLE_PIN      GPIO_NUM_22  // labeled SpinEnbl

#define MIST_PIN                GPIO_NUM_21  // labeled Mist
#define FLOOD_PIN               GPIO_NUM_25  // labeled Flood
#define PROBE_PIN               GPIO_NUM_32  // labeled Probe

#define CONTROL_SAFETY_DOOR_PIN GPIO_NUM_35  // labeled Door,  needs external pullup
#define CONTROL_RESET_PIN       GPIO_NUM_34  // labeled Reset, needs external pullup
#define CONTROL_FEED_HOLD_PIN   GPIO_NUM_36  // labeled Hold,  needs external pullup
#define CONTROL_CYCLE_START_PIN GPIO_NUM_39  // labeled Start, needs external pullup
*/
`

bdring, 2020-05-31:
Thanks...easy fix when we get the info.

you cannot use pins 34-39 without attached hardware with pullup resistors. A floating I/O pin has told Grbl that your door is open.

Comment out all these pins.

```
#define CONTROL_SAFETY_DOOR_PIN GPIO_NUM_35 // labeled Door, needs external pullup
#define CONTROL_RESET_PIN GPIO_NUM_34 // labeled Reset, needs external pullup
#define CONTROL_FEED_HOLD_PIN GPIO_NUM_36 // labeled Hold, needs external pullup
#define CONTROL_CYCLE_START_PIN GPIO_NUM_39 // labeled Start, needs external pullup
```

Please consider making a donation to this project.

wayneosdias, 2020-05-31:
Perfect, thanks. Thanks also for your work on the psoc grbl port too :) Do you want me to donate or would you prefer I buy your product on Tindie?

closed 2020-05-31 by wayneosdias