bdring/Grbl_Esp32 #423: Homing not working after doing all the basic testing steps

dakoner, 2020-06-05:
I have a simple 2-axis machine with limit switches at the most negative position.  I'm stuck trying to get homing to work.  I have the 3-wire switches shown in https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-Limit-Homing-Switch with the two wires connected (signal and ground).  I've verified that the X and Y limit switches report Pn:X and Pn:Y when they are pressed, and don't when they are released, and that they are connected to the correct axes.

When I try to home, each axis does the same thing: makes two small steps in the positive direction at the fast search rate for a distance of $27 then stops.  No error is reported (unless the limit switch is already pressed, in which case it does ALARM: 8 ALARM: 3).

I've configured my machine with:

#define LIMIT_MASK              B011
#define X_LIMIT_PIN             GPIO_NUM_17
#define Y_LIMIT_PIN             GPIO_NUM_4

#ifdef HOMING_CYCLE_0
    #undef HOMING_CYCLE_0
#endif
#define HOMING_CYCLE_0 (1<<X_AXIS) // X first

#ifdef HOMING_CYCLE_1
    #undef HOMING_CYCLE_1
#endif
#define HOMING_CYCLE_1 (1<<Y_AXIS)

#ifdef HOMING_CYCLE_2
    #undef HOMING_CYCLE_2
#endif

and runtime settings:
$3=0
$5=1
$20=0        
$21=0  
$22=1        
$23=3        
$24=200.000  
$25=2000.000 
$26=250      
$27=1.000  

all of which seem to make sense.

bdring, 2020-06-05:
Appears a limit switch is reporting as touched and won't clear. Send a ? Command and post the response

dakoner, 2020-06-05:
The limit switches are clear when not pressed:
<Idle|MPos:-299.000,-299.000,0.000|FS:0,0>

X limit switch pressed:
<Idle|MPos:-299.000,-299.000,0.000|FS:0,0|Pn:X>

Y limit switch pressed:
<Idle|MPos:-299.000,-299.000,0.000|FS:0,0|Pn:Y>

Here's what I see if I send $H then hit ? repeatedly:

Grbl 1.2a ['$' for help]
[MSG:'$H'|'$X' to unlock]
$H
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Home|MPos:-299.000,1.000,0.000|FS:0,0|Ov:100,100,100>
<Home|MPos:-299.000,0.000,0.000|FS:0,0>
<Home|MPos:-299.000,0.000,0.000|FS:0,0>
<Home|MPos:-299.000,0.870,0.000|FS:0,0>
<Home|MPos:-299.000,1.000,0.000|FS:0,0>
ok
ok
<Idle|MPos:-299.000,-299.000,0.000|FS:0,0>

I'm pretty sure the system is working at the physical level, as I had this working with GRBL and a Nano previously.

bdring, 2020-06-05:
Strange, from your status reports, It looks like it did a successful home.

Are you able to jog the machine in both directions on each axis?

dakoner, 2020-06-05:
Yes, both axes jog fine in the Web UI.  I also enabled software debounce, didn't make any difference.

If I unplug the endstop signal wires from the ESP32, I don't see Pn:X or Pn:Y, and homing runs until timeout.  

I double checked the (normally open) switches and they seem to have an OK signal (I don't have an oscilloscope to test, just a multimeter).

bdring, 2020-06-05:
Can you post your full machine definition file and the startup text?

bdring, 2020-06-05:
You could also try turning hard limits on. Then try jogging around. It will alarm if a limit is detected due to noise.

dakoner, 2020-06-05:
I tried enabling hard limits and that immediately reports:
Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.

I assume this means that the pins are flickering constantly, faster than the once-per-second status reporting.  Let me do some more basic testing.

bdring, 2020-06-05:
I am going to close this issue and give you [a link to our slack](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA).

closed 2020-06-05 by bdring