bdring/Grbl_Esp32 #426: Dual motor support for self-squaring gantry homing.

master-kill, 2020-06-06:
Hello, I used grbl with the Arduino Nano controller before. Now I want to use esp32, but there are some difficulties.
I have a 2-axis laser cnc. on the y axis I have two motors.
the problem is that returning to the square does not work. for this you need to have your own switch for each motor (Y and Y2)
This feature works great on Arduino Nano.
it should work like this:
1 all axles start driving home (X, Y, Y2)
2 each axis (X, Y, Y2) stops near its switch (switch X, switch Y, switch Y2)
3 execution of N_HOMING_LOCATE_CYCLE
4 end (cnc in home position and square)

the problem is that after lim Y is triggered, motor Y2 stops and the machine does not become square. then N_HOMING_LOCATE_CYCLE is executed, but the machine is not square :(

How to enable lim Y2 in the configuration of the machine?

[image]
[image]

bdring, 2020-06-06:
Try $HY to home just the Y. I am not sure you can use auto square with 2 axes at once.

michael-ring, 2020-07-19:
I have a similar configuration, homing works just fine when you only use only Y_LIMIT_PIN and have the two homing switches connected together so that they work as an 'OR', which is easy with mechanical switches, simply connect them in parallel.

Here's the relevant part of my config where I use the Z Axis as Y2:

#define Y_STEP_PIN              GPIO_NUM_25
#define Y_DIRECTION_PIN         GPIO_NUM_27

#define Y2_STEP_PIN             GPIO_NUM_17
#define Y2_DIRECTION_PIN        GPIO_NUM_14
//#define Z_STEP_PIN              GPIO_NUM_17
//#define Z_DIRECTION_PIN         GPIO_NUM_14

#define X_LIMIT_PIN             GPIO_NUM_13
#define Y_LIMIT_PIN             GPIO_NUM_5
//#define Z_LIMIT_PIN             GPIO_NUM_19
#define LIMIT_MASK              B111

#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)
#undef HOMING_CYCLE_2

#define USE_GANGED_AXES
#define Y_AXIS_SQUARING

It would however be easier if it were possible to define an Y2_LIMIT_PIN that is different from Y_LIMIT_PIN and then have GRBL do the logical 'OR'. 

Reason is that when you use optical endswitches you will need two diodes and an resistor to build this 'or' which is a (small) pain in the ass to build when you do not have the components at hand. 

I ended up designing a small PCB that plugs on the CNC-Shield.

Reading through my config I am asking myself why I did this:
#define LIMIT_MASK              B111

I think that should be enough:
#define LIMIT_MASK              B11

Romain3481, 2021-04-12:
Hi ,
I would like an Y2 endstop to squaring too !
I read the wiki page on squaring, but i have a free pin to add an Y2 endstop, that seems more logical than "squaring in 3 times".
(Thank you so much to contributors for this firmware !)