bdring/Grbl_Esp32 #429: Y axis not at -max

JensHauser, 2020-06-09:
Hi,

installed GRBL ESP32 for the first time on a self-made 3 axis router.
Homing work like a charm.

There is just one thing I don`t understand.

When all home positions are reached, the web interface shows me the following values

	Xm:-5.000 (-1000mm max.)
	Ym:-420.000 (-420mm max.) <---- issue
	Zm:-5.000 (-120mm max.)

X and Z are fine, but why is Y at -420mm and not at -5mm?

If I use G code to go to absolute values X0 Y0 Z0, only Y goes to the other end.

Below you´ll find the configuration I consider to be useful to understand the environment.

Greetings from Germany,
 Jens

1.
Sources compiled ander Arduino IDE, just one adjustment in machine.h
#include "Machines/3axis_v4.h"

2. Versions
UI: 2.1b61 / FW: 1.1f (20200419)

3. GRBL parameters
$0=10
$1=250
$2=0
$3=2
$4=1
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=2
$24=200.000
$25=2000.000
$26=44
$27=5.000
$30=1000.000
$31=0.000
$32=0
$100=160.000
$101=160.000
$102=160.000
$110=4000.000
$111=4000.000
$112=1500.000
$120=50.000
$121=50.000
$122=50.000
$130=1000.000
$131=425.000
$132=120.000

bdring, 2020-06-09:
That is likely due to homing direction and Grbl homes in negative space.

It should not matter. You will be zeroing the work coordinates anyway.

JensHauser, 2020-06-09:
Thanks for your advice, I‘m on the way to learn about these coordinate system.

Do you see a chance to flip Y axis somewhere in the sources or so to ease my steps into CNC a little bit?

MitchBradley, 2020-06-10:
You can change the direction of the Y axis with the $3 setting - direction invert mask.  It is explained in the normal GRBL documentation https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration or the pocket guide at http://www.diymachining.com/downloads/GRBL_Settings_Pocket_Guide_Rev_B.pdf .  Grbl_Esp32 

I see that your $3 setting is "2".  As shown in the pocket guide, "2" means "reverse Y axis".  But instead of just taking my word for it, you should read those documents carefully and experiment with the various settings, so you will come to understand them.  Then you will be able to solve any other configuration problems as they occur.

JensHauser, 2020-06-11:
Thank you so much. That gave me orientation about this issue :-)

closed 2020-06-11 by JensHauser

JensHauser, 2020-06-22:
A follow-up on this issue with some thoughts I want to share.
After some time of trying and learning, I make 3D models with Autodesk Fusion 360 and generate G code. Setting the XYZ axes is a one-time task, including +/- direction and individually setting the model and raw material coordinate origin.
So don't bother with this issue, it happens once only. The rest is handled via coordinate systems in G code and abstracted via software.