bdring/Grbl_Esp32 #431: PWM output missing due to SPINDLE_ENABLE reversed.

dart1680, 2020-06-13:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?---**-YES**

What version of the Arduino IDE are you using? ....**1.8.12**

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? ....**1.0.4**

Are you using the master branch of Grbl_ESP32?....**.yes**

Have you made any edits or configuration changes (list them) to the  firmware?
edit SSID & PASSWORD in config.h

#include "Machines/lowrider_v1p2.h" in machine.h

edit lowrider_v1p2.h in \Machines to read 

```#ifdef USE_SPINDLE_RELAY
    #define SPINDLE_TYPE SPINDLE_TYPE_RELAY
    #define SPINDLE_OUTPUT_PIN GPIO_NUM_2
#else
    #define SPINDLE_TYPE SPINDLE_TYPE_PWM
    #define SPINDLE_OUTPUT_PIN         GPIO_NUM_16

    #define SPINDLE_ENABLE_PIN      GPIO_NUM_32
#endif

// Note: Only uncomment this if USE_SPINDLE_RELAY is commented out.
// Relay can be used for spindle or either coolant
#define COOLANT_FLOOD_PIN     GPIO_NUM_2
//#define COOLANT_MIST_PIN      GPIO_NUM_2

#define X_LIMIT_PIN             GPIO_NUM_17
#define Y_LIMIT_PIN             GPIO_NUM_4
#define Z_LIMIT_PIN             GPIO_NUM_15
#define LIMIT_MASK              B111

#ifndef ENABLE_SOFTWARE_DEBOUNCE   // V1P2 does not have R/C filters
    #define ENABLE_SOFTWARE_DEBOUNCE
#endif

#define PROBE_PIN               GPIO_NUM_35

// The default value in config.h is wrong for this controller
#ifdef INVERT_CONTROL_PIN_MASK
    #undef INVERT_CONTROL_PIN_MASK
#endif

#define INVERT_CONTROL_PIN_MASK B1110

#define CONTROL_RESET_PIN           GPIO_NUM_34  // needs external pullup
#define CONTROL_FEED_HOLD_PIN       GPIO_NUM_36  // needs external pullup
#define CONTROL_CYCLE_START_PIN     GPIO_NUM_39  // needs external pullup

#define DEFAULT_STEP_PULSE_MICROSECONDS     3
#define DEFAULT_STEPPER_IDLE_LOCK_TIME      255 //  255 = Keep steppers on

#define DEFAULT_STEPPING_INVERT_MASK    0 // uint8_t
#define DEFAULT_DIRECTION_INVERT_MASK   0 // uint8_t
#define DEFAULT_INVERT_ST_ENABLE        0 // boolean
#define DEFAULT_INVERT_LIMIT_PINS       1 // boolean
#define DEFAULT_INVERT_PROBE_PIN        0 // boolean

#define DEFAULT_STATUS_REPORT_MASK      1

#define DEFAULT_JUNCTION_DEVIATION  0.01 // mm
#define DEFAULT_ARC_TOLERANCE       0.002 // mm
#define DEFAULT_REPORT_INCHES       0 // false

#define DEFAULT_SOFT_LIMIT_ENABLE 0 // false
#define DEFAULT_HARD_LIMIT_ENABLE 0  // false

#define DEFAULT_HOMING_ENABLE           1  // false
#define DEFAULT_HOMING_DIR_MASK         3 // move positive dir Z,negative X,Y
#define DEFAULT_HOMING_FEED_RATE        100.0 // mm/min
#define DEFAULT_HOMING_SEEK_RATE        200.0 // mm/min
#define DEFAULT_HOMING_DEBOUNCE_DELAY   250 // msec (0-65k)
#define DEFAULT_HOMING_PULLOFF          2.0 // mm

#ifdef USE_SPINDLE_RELAY
    #define DEFAULT_SPINDLE_RPM_MAX 1.0 // must be 1 so PWM duty is alway 100% to prevent relay damage
#else
    #define DEFAULT_SPINDLE_RPM_MAX 1000.0 // can be change to your spindle max
#endif

#define DEFAULT_SPINDLE_RPM_MIN 0.0 // rpm

#define DEFAULT_LASER_MODE 0 // false

#define DEFAULT_X_STEPS_PER_MM 200.0
#define DEFAULT_Y_STEPS_PER_MM 200.0
#define DEFAULT_Z_STEPS_PER_MM 800.0

#define DEFAULT_X_MAX_RATE 8000.0 // mm/min
#define DEFAULT_Y_MAX_RATE 8000.0 // mm/min
#define DEFAULT_Z_MAX_RATE 3000.0 // mm/min

#define DEFAULT_X_ACCELERATION (200.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
#define DEFAULT_Y_ACCELERATION (200.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
#define DEFAULT_Z_ACCELERATION (100.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2

#define DEFAULT_X_MAX_TRAVEL 500.0 // mm NOTE: Must be a positive value.
#define DEFAULT_Y_MAX_TRAVEL 500.0 // mm NOTE: Must be a positive value.
#define DEFAULT_Z_MAX_TRAVEL 80.0 // mm NOTE: Must be a positive value.
```
The code compiles but there is no PWM output. The signal SPINDLE_ENABLE is LOW when enabled with an M3 command  and HIGH when switched off with a M5 command (which is backwards) so the PWM signal does not make it through the 3 to 5v level changer. If I uncomment
``` // #define INVERT_SPINDLE_ENABLE_PIN // Default disabled. Uncomment to enable. 
```
then the SPINDLE_ENABLE signal changes to active high  but the PWM output also changes from active HIGH  to  active LOW. There does not appear to be an INVERT PWM option.

Sorry if this is in the wrong place. Feel free to move/delete

MitchBradley, 2020-06-13:
Try this fix https://github.com/bdring/Grbl_Esp32/tree/Fix431 and see if it does what you want.  It adds a separate config options  INVERT_SPINDLE_OUTPUT_PIN in addition to the existing INVERT_SPINDLE_ENABLE_PIN .

MitchBradley, 2020-06-13:
The patch is https://github.com/bdring/Grbl_Esp32/commit/66ac0eaa425d8329d94d1b5a5580940b5a340fe7 if you  would just like to apply that to your tree.

dart1680, 2020-06-14:
OK, I tried the patch and it adds the option to reverse the pwm output pin function. If I enable both spindle enable invert AND spindle PWM invert the pwm signal is still wrong but now you have added the OPTION if I leave the invert spindle output pin commented out it all works correctly. ie
```
// Inverts the spindle enable pin from low-disabled/high-enabled to low-enabled/high-disabled. Useful
// for some pre-built electronic boards.
#define INVERT_SPINDLE_ENABLE_PIN // Default disabled. Uncomment to enable.

// Inverts the spindle PWM output pin from low-disabled/high-enabled to low-enabled/high-disabled.
// #define INVERT_SPINDLE_OUTPUT_PIN // Default disabled. Uncomment to enable.
```
has enable going high with an M3 and the PWM pulsing high
The nomenclature  seems all ass-about-face to me but it works.

Thanks

MitchBradley, 2020-06-14:
Propose a different name and write the instructions then.

dart1680, 2020-06-14:
Sorry Mitch, that maybe came across wrong, the existing script states that both the SPINDLE_ENABLE_PIN and the SPINDLE_OUTPUT_PIN should be active HIGH and in my case that is not what is happening. I did not mean to offend and do appreciate all the effort both you and Bart put in, nothing *needs* to be re-written as it now works 'ass-about-face' and I would not know how to do so anyway, I just thought as this is your baby you might have been more interested in the root cause rather than just writing a fix.
Thank you for your efforts.

commit 66ac0eaa (2020-06-14): Fix #431

Added INVERT_SPINDLE_OUTPUT_PIN option
closed 2020-07-20 by bdring