bdring/Grbl_Esp32 #433: No motor drive, intermittant connection errors.....unless connected via USB as well as wifi?

dart1680, 2020-06-13:


bdring, 2020-06-13:
Need more info.

Describe all  the hardware you are using and paste the startup text in a reply.

dart1680, 2020-06-14:
Thanks for your reply Bart,

Hardware -
ESP32s
PCB by JLCPCB from your gerbers
3x drv8825
3xNEMA17's

Startup text from Arduino serial monitor -
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

[MSG:Grbl_ESP32 Ver 1.2a Date 20200514]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:LOWRIDER_V1P2]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:PWM spindle Output:16, Enbl:32, Dir:-1, Freq:5000Hz, Res:13bits]

[MSG:Client Started]
[MSG:Connecting Orlando]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connected with 192.168.1.14]
[MSG:Connected with 192.168.1.14]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]

Grbl 1.2a ['$' for help]
[MSG:'$H'|'$X' to unlock]

Everything seems good at this stage, I can refresh the TF card contents, I
can obtain GRBL settings and obtain ESP3D status all over wifi but

After clicking on the jog Y button I get this....

<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:740.280,-50.030,3.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|WCO:740.280,-50.030,3.000>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
$X
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
ok
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:740.280,-50.030,3.000>
[... 74 lines cut ...]
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
$J=G91 G21 F1000 Y-10
Connection error

This occulance it caused the connection error and the page has to be reset
(F5) to re-commence comms. Sometimes the verbose window just freezes for a
few seconds....maybe 20 or so and then the messages continue. At no time
does any stepper motor engage.

The ESP32 is  already connected over USB as I watch  the serial comms with
Arduino serial monitor but the problem is the same if I leave the USB
disconnected). If I then open up Universal Gcode Sender and connect over
USB I can operate the ESP32 normally both from UGS and over the wifi link
simultaneously without problem.

Attached ESP3D status screenshot and the UI just after startup before
attempting to move any steppers.

Thankyou for your help

Regards

Mike

dart1680, 2020-06-14:
Bit of extra info, once the usb connection has been made and a $X sent I can disconnect the USB and the wifi continues to work OK but if I close the USB connection before unplugging it the wifi stops working too.
This is starting to look to my un-informed eye like the problem Android has with rj45 and wifi network connection priority

dart1680, 2020-06-14:
This cannot be right...there must be plenty of people out there with ESP32s powered lowriders right?....My dev board came from Aliexpress with a CH340G USB to serial chip and I notice from https://github.com/bdring/Grbl_Esp32/issues/410 that chip employs a CP2102...could this be significant?...there must be others using ESP32s's with CH340G's surely??

OK...I'll shut up now and wait for y'all to get up out of bed :-)

MitchBradley, 2020-06-14:
I would investigate two things with a scope or logic analyzer:
a) The quality of the 5V and 3V3 power at the module pins during the events that trigger the problems.  If the power is dipping or glitching, you might need to add extra bulk capacitance, so a 100uF or 220uF capacitor at the 5V input.
b) The behavior of the EN line when the problem happens.  If EN is going low, you need to look at the RTS and DTR lines from the CH340 to see if anything funny is happening there.  In general, I have had better experience with CP2102 than with CH340.  I think that, at least, the CP2102 driver software is better, because I have seen CH340 drivers crash Macs.

MitchBradley, 2020-06-14:
Pretty much every ESP32 and ESP8266 dev board has a cross-coupled two-transistor latch circuit between RTS/DTR and EN/A0, for the purpose of putting the ESP into boot mode under USB control.  If that is messed up, or if RTS or DTR is glitching, it could be resetting the ESP.

dart1680, 2020-06-14:
Thank for your suggestions, I will  have a look at those points - not sure though how that sits with it all working perfectly if a USB serial connection is opened first

MitchBradley, 2020-06-14:
> not sure though how that sits with it all working perfectly if a USB serial connection is opened first

RTS DTR state

dart1680, 2020-06-14:
OK...well that didn't go well. I could not see anything untoward on the power rails with what I have here, however, connecting  100uF caps on the 3v3 and 5v rails at the esp32 module did stuff one of my ESP32s's. It started by continually resetting and now it will no longer enter boot mode....so that is it for me, I've been at this for over 80 hours....... I give up, I'm going back to an 8 bit solution, I know that works.  Thankyou for all your input but all this 'still in development' stuff is not for me. 

"RTS DTR state"....... how does that change after connecting and then disconnecting via usb?   The environment hasn't changed.   you are most likely right but I don't understand it, so it is not for me.

dart1680, 2020-06-14:
closing the issue

MitchBradley, 2020-06-14:
Note that I said 100uF on the 5V, not on the 3V.  Adding that much capacitance on 3V could overstress the linear regulator.

RTS and DTR are outputs from the USB serial converter.  Connecting/disconnecting the USB is very likely to change their state.  The details depend on the USB serial chip, its driver, and how those lines are connected to the cross-coupled latch circuit that drives the EN and GPIO0 pins.

HuubBuis, 2020-06-27:
I have a similar problem on my esp32 dev board version 1. It started today after functioning ok for about 4 days. The serial reports a problem reading the flash at address 0x01000 (not sure about the 000 in the address) and the device reboots. It has gone from once every 5 minutes to not able to connect any more.  Finally I disabled the WiFi in code and the connection problems are gone. Sadly I haven't configured bluetooth using the web ui so I can't test if bluetooth has the same problem.
I have tried to solve it using a external PSU on the +5V, 3V3, 100uF cap (low esr) on +5V and 3V3 but that didn't help. 
It could be the room temperature that has increased for 20° to 30° (summer now) and a problem in the internal power regulator of the ESP32. I have read some treads that when WiFi starts, the sudden power requests causes the problem.
I have ordered a new module but that will take some time to arrive. In the meanwhile, I continue testing using the serial connection. At the end of this week, the temperature will lower to "normal" levels. I will also test by cooling the ESP32 using some ice cubes.  I will report these results.

dart1680, 2020-06-29:
I can confirm that a new ESP32 dev board with the CP2102 works fine out of the box and the two CH340G equipped ESP32 dev boards didn't.    Caveat emptor.

closed 2020-06-29 by bdring

HuubBuis, 2020-06-29:
I have tested cooling the ESP32 but that didn't make a difference.  Today, it is about 5° cooler now and ESP32 WiFi is working again but now I have an other issue,  Grbl raises an error when sending $$ after sending some other commands. Maye by board is toast!
For the next board I make, I will power the ESP32 by 7 volts and and use a 470uF capacitor. That will give more margin to handle the power surge when WiFi starts.
I will also order a CP2102 board to see if it does a better job.

MitchBradley, 2020-06-29:
If you power with 7V, be careful about cooling the linear regulator that drops the voltage to 3.3V.  The power that regulator dissipates will more than double with 7V input compared to 5V input.  (7-3.3)/(5-3.3) = 2.18

HuubBuis, 2020-06-29:
According to a dev kit guide I have read, it could be powered from 5..12 Volts.
The current module draws about 0.2 A. The power dissipation for the 3v3 converter will be about 6W. That will give a temperature rise from at least 90 °C. 
You are right, this is way to much. I will measure the current module temperature to see how high i can go!
Thanks!

MitchBradley, 2020-06-29:
Another thing just came to mind - There are some options about how the FLASH is powered, and others for its mode (qio, dio, etc) and clock frequency. Since your problems seem vaguely related to FLASH access, maybe there could be an issue with the various FLASH modes.  Sorry I can't give you exact instructions for changing those settings, as I do not have the information on the tip of my tongue.  I could look them up, but I presume that you can too.

MitchBradley, 2020-06-29:
At 0.2A and 5V input and 0.2, the converter will dissipate (5-3.3)*0.2 = 0.34W.  At 7V, it's (7-3.3)*0.2 = 0.74W.  To get to 6W at 0.2A, the input voltage would need to be 33V.  To get 6W at 12V input, the current would need to be 0.69A.

HuubBuis, 2020-06-29:
> 
> 
> At 0.2A and 5V input and 0.2, the converter will dissipate (5-3.3)*0.2 = 0.34W. At 7V, it's (7-3.3)*0.2 = 0.74W. To get to 6W at 0.2A, the input voltage would need to be 33V. To get 6W at 12V input, the current would need to be 0.69A.

You dissipation calculations are right. The 7 volts input can probably done. I will power the module using my lab psu and measure the temperature to be sure because the temperature rise depends greatly upon the design (size) of the pad. A will do this when I have made the enclosure for the controller, somewhere end of this week.

> Another thing just came to mind - There are some options about how the FLASH is powered, and others for its mode (qio, dio, etc) and clock frequency. Since your problems seem vaguely related to [...]

The flash frequency and mode can be set in the Arduino IDE.  How to set the flash voltage is described in one of the espressif docs but I don't remember how, i am getting old! I have only 3 docs and will read them again.

For now i will use Bluetooth so i can go on testing. When the new module arrives, I decide if it is worth trying to get this one working on WiFi or use it some day for a stress test to see what the limits are.

HuubBuis, 2020-06-30:
I have tested my board using all the combinations of the two selectable flash speeds (40,80 MHz) and flash modes (DIO,QIO, DOUT,QOUT) but it didn't cure the problem.
I have read the docs again for the flash voltage. 
`For ESP32 chips that contain an embedded flash, users need to note the logic level of MTDI. For example, ESP32D2WD contains an embedded flash that operates at 1.8 V, therefore, the MTDI should be pulled high. ESP32U4WDH contains an embedded flash that operates at 3.3 V, therefore, the MTDI should be low.
`
My ESP32 board has 4MB embedded flash, so nothing to setup!
I hope the new board will do a better job!