bdring/Grbl_Esp32 #437: Best option for long cables (board - computer)

vitalibr, 2020-06-15:
has anyone ever needed to connect the controller board to the pc with a 5 meter cable (or even longer)?  I know I will have problems with USB, I thought about using Ethernet, or I could use Rs485 (I don't know if it makes sense).

my CNC mill is big and the control panel will need a long cable to connect to my computer (raspberry)

just-jason, 2020-06-15:
Wifi or Bluetooth ? Just an idea

vitalibr, 2020-06-15:
> Wifi or Bluetooth ? Just an idea

I thought about it, but I don't know if a wireless connection is safe for a machine that can be used for industrial environments

ningzhang-github, 2020-06-16:
You could consider one of the field buses. For example, CANopen, Modbus or even EtherCAT. 
https://en.wikipedia.org/wiki/Fieldbus
https://en.wikipedia.org/wiki/CANopen
https://en.wikipedia.org/wiki/Modbus
https://en.wikipedia.org/wiki/EtherCAT

closed 2020-06-19 by vitalibr