bdring/Grbl_Esp32 #441: Issue with Trinamic drivers and motors not responding

rdh9616, 2020-06-16:
After updating and recompiling the firmware my Trinamic drivers are no longer working. There were some structural changes to the Grbl_Esp32 source code and I am not sure if that is what is causing the issue.

Here is the output from the serial monitor when booting:

[MSG:Grbl_ESP32 Ver 1.2a Date 20200613]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_TMC2130_PEN V2]
[MSG:TMCStepper Init using Library Ver 0x000602]
[MSG:X Trinamic driver test failed. Check connection]
[MSG:Y Trinamic driver test failed. Check connection]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Z Servo range 0.000 to 5.000]
[MSG:No spindle]

Any assistance is appreciated!

Thank you,

Ray

KingMo5h, 2020-06-17:
Wich board you are using?
Everything connected right?
Power supply on for the drivers?

bdring, 2020-06-17:
I'll look at this later today. 

Do you have main motor power on before you start the ESP32?
Are you sure the drivers in SPI mode?

rdh9616, 2020-06-18:
I thick it was an issue with the board as I flashed the firmware to another board, both official devkit boards, and everything is working. Closing the issue.

closed 2020-06-18 by rdh9616