bdring/Grbl_Esp32 #449: Lot's of new Grbl_ESP32 features!

bdring, 2020-06-18:
After several weeks of development by @bdring @MitchBradley and @odaki, we have pushed some major changes to the [Devt](https://github.com/bdring/Grbl_Esp32/tree/Devt) branch. We invite everyone to test it and leave feedback on this thread or on Slack.

### I2S Shift Register Chip Support (@odaki @MitchBradley @bdring )

The ESP32 has just enough pins for an average CNC machine. If you wanted to go with lots of motors and features you were out out of luck. I2S Shift registers are an easy and cheap way to additional output pins. The development board we used had 4 chips, adding an additional 32 output pins. There is a [github repo for the dev board](https://github.com/bdring/6-Pack_CNC_Controller) if you want to see more details.

[image]

### New Settings System (@MitchBradley )

The $$ setting system was always problem. The numbers were limiting and hard to remember. It was also hard to add new settings without corrupting your old settings. The new system uses easily an readable text format.

The system was rewritten from the ground up. It still has basic backward compatibility with the original Grbl settings to keep gcode senders happy. New setting like $StallGuardDebugMask will only be available in the $text type format. Here is a partial list you get by sending $S

```
$SpindleMaxPWM=100.000
$xStepsPerMm=200.000
$yStepsPerMm=200.000
$zStepsPerMm=100.000
$xMaxRate=5000.000
$yMaxRate=1000.000
$zMaxRate=1000.000
$xAcceleration=20.000
$yAcceleration=200.000
$zAcceleration=200.000
$xMaxTravel=100.000
$yMaxTravel=300.000
$zMaxTravel=100.000
$xRunCurrent=0.800
$yRunCurrent=0.500
$zRunCurrent=0.500
$xHoldCurrent=0.250
$yHoldCurrent=0.250
$zHoldCurrent=0.250
$xMicrosteps=16
$yMicrosteps=32
$zMicrosteps=16
$xStallGuard=8
$yStallGuard=16
$zStallGuard=16
```
You can send $L to see how the numbers vs text match up. Here is a partial list.

```
$0 => $StepPulse
$1 => $StepperIdleTime
$2 => $StepInvertMask
$3 => $DirInvertMask
$4 => $StepEnableInvert
$5 => $LimitInvert
$6 => $ProbeInvert
$10 => $StatusMask
$11 => $JunctionDeviation
$12 => $ArcTolerance
$13 => $ReportInches
$20 => $SoftLimits
$21 => $HardLimits
$22 => $HomingEnable
```
Send $E to see what the error numbers mean. Partial list.

```
0: No error
1: Expected GCodecommand letter
2: Bad GCode number format
3: Invalid $ statement
4: Negative value
5: Setting disabled
6: Step pulse too short
7: Failed to read settings
8: Command requires idle state
9: GCode cannot be executed in lock or alarm state
10: Soft limit error
11: Line too long
12: Max step rate exceeded
13: Check door
14: Startup line too long
15: Max travel exceeded during jog
16: Invalid jog command
17: Laser mode requires PWM output
```
There are many more usability features too.

**Note:** We plan to do a little name tweaking and add a hierarchical tree like structure to them.

### Motor Classes (@bdring )

Grbl_ESP32 supports a lot more motor types than standard Grbl including simple step/dir types, Trinamic SPI types, RC servo and Unipolar stepper motors. This was bloating and making a mess of the core motion controller code. Now the motors use object oriented c++ code. This simplifies things and gives a huge reduction in code.  Additional motor type will be easy to add.

MitchBradley, 2020-06-18:
The I2S dev board is undergoing some important revisions, so if you are considering building one, it might be best to wait.

happytm, 2020-06-18:
@MitchBradley  I hope there is option for couple of  brushed DC motors too with encoder input option like linked below: 

https://github.com/misan/dcservo

Thanks.

MitchBradley, 2020-06-18:
The new board supports plug-in modules for spindles and inputs, and can send the step and direction pins to output connectors via pololu-socket bridging adapters.  That should suffice for the hardware side of what you want.  It won't solve the major problem which is software, so this thread is not the right place to reiterate the request for brushed servo support.

happytm, 2020-06-18:
@MitchBradley Thank you.

RootCNC, 2020-06-19:
Hi Gents, 

love the project and I can see you're doing some great work bring new hardware out. I was wondering If I could lend a hand with the hardware design? As part of my Root CNC project, I’ve been looking to design a new controller for one of my machines (Root 4). I’ve looked at MESA cards, GRBL, Parallel cards, MACH, Arduino based project and either they’re too expensive or limited on features and I think this project can sit nicely within this area and offers feature rich options with a low buy-in cost.
I’m currently looking at designing a controller with the focus of powering a feature rich Root CNC but whilst designing this board It might be a nice idea to incorporate more features you were thinking of. I’m be happy to help design boards as I am an electronic engineer by profession. Just a thought – happy lend a hand with this great project.

bdring, 2020-06-19:
@RootCNC The design I am working on is detailed in the first post of this issue. It works great with no problems, but I am looking at changing the input section to be modular. The spindle modules worked well, so I am thinking about extending that concept to inputs. I put my current thoughts on [this wiki page](https://github.com/bdring/6-Pack_CNC_Controller/wiki/Input-Modules).

RootCNC, 2020-06-19:
@bdring ah I see (still new to GitHub). if you need a hand, please let me know. I've got a scheme planned for an fully isolated board (assuming external motor drivers are used) which might be of use. 

Keep up the good work.

vitalibr, 2020-06-19:
Awesome!

Can I put the Spindle RS485 module in "Spindle Module 1" or "Spindle Module 2"? Or will it be mandatory to be used in one of the slots?

And $StallGuardDebugMask is amazing! great idea!

bdring, 2020-06-19:
@vitalibr  What slot you use depends on the design of the controller motherboard. RS485 needs some bi-directional GPIO. The Relay only needs 1 output. 

In my current design I have GPIO going to one and just outputs going to the other. Therefore the second is limited in what modules can be used. I may change a few things on the final design. The primary purpose was just to test new features.

Note: Grbl only supports one spindle. The others would be used for coolant, vacuum, tool changer, etc. The term spindle module will probably change to something like "Output Module".

@happytm There is no reason you could not put a motor controller on an Output Module. You could make a small, smart, brushless DC controller that accepted step/dir/enable and possibly a fault return. Like @MitchBradley said, please don't do that on this thread.

happytm, 2020-06-19:
@bdring Thank you.

MitchBradley, 2020-06-20:
@RootCNC Could you supply a list of the signals you need for your Root 4 machine?  We could compare it to what we will have available on the new modular board.  If there are gaps maybe we could come up with plug-in module ideas to fill those gaps, and perhaps you could design some modules.

RootCNC, 2020-06-24:
@MitchBradley sure thing; so I've put together this repo (I thought this was the best way share some ideas) 

[ESP32-GRBL-Root-Controller](https://github.com/RootCNC/ESP32-GRBL-Root-Controller )

Its very much a concept but the top level SVG might be a good idea to help identify my thoughts. link - [Here](https://github.com/RootCNC/ESP32-GRBL-Root-Controller/blob/master/Concept_Idea.svg). The block diagram also has some preliminary parts defined in it, to support the concept. 

Pin allocation can be found in the excel spreadsheet [here](https://github.com/RootCNC/ESP32-GRBL-Root-Controller/blob/master/ESP32_ROOT_4_PINOUT.xlsx).

let me know your thoughts :)

MitchBradley, 2020-06-24:
Okay, so it looks like Bart's new "6-pack" board will handle everything in your use case:
* The 6 channels of stepping output is done via I2S (not I2C) shift registers - 74AHCT595 - that can drive 5V optocouplers.  There are bridging modules for the Pololu sockets so you can connect external drivers easily.  The I2S stepping scheme is completely jitter-free at 250 kHz edge rate, 125 kHz pulse rate.
* There are 5 plug-in slots for I/O modules.  They all have 4 common pins that can be used for an I2C expander on the module, and each has 4 non-common GPIO pins that can be used for dumb I/O or anything you wish to configure the ESP to drive.
*  Your spindle card could be designed as a module, if one of Bart's existing spindle modules is not already suitable (e.g. not isolated well enough).
* Another one of Bart's modules is a 4-channel "bulletproof" input module, with optocoupled, RC-filtered, Schmitt trigger buffers.  If you need more inputs you can use more than one of these modules.
* If you need even more custom I/O, you can make a module with an I/O expander or even a small microprocessor like an STM32F030.  I am designing some "smart modules" like this, so they may be available off the shelf in the not too distant future.

RootCNC, 2020-06-27:
Nicely said. I'll look more into the shift registers you've used. With a quick calculation the step pulses needed for a Root CNC to traverse at a high speed is roughly 50KHz step speed that seems a comfortable speed for the machine, Although higher steps around 75KHz can be used, but the machine does sounds a tad UN-happy. 

Have you guys assess the number of shift register than can be daisy changed whilst still producing a good output rate for motion control? 

I'm still looking at producing a Root CNC focused controller using the software you and the team have put together (as It look like it fits the bill perfectly). I'll take what you have said on board and keep you posted on the progress.

MitchBradley, 2020-06-27:
I think it would be possible to daisy chain 8 shift registers at 250 kHz edge rate/125 kHz step rate.

closed 2020-08-23 by bdring