bdring/Grbl_Esp32 #451: Relay to switch milling motor on/off

JensHauser, 2020-06-22:
Hi folks,

my self-developed board works with 4 axes (plus limit switches, feed, hold, reset) and should switch on/off a milling motor driven by a relay. No PWM, the speed is adjusted manually with a slider directly at the motor.

So I prepared to use GPIO35 (Door/Flood/Mist are not in use), but serial Arduino interface says "no spindle pin".

I played with many variations of Spindle PWM, enagle, etc., but no luck.

In the [docs](https://github.com/bdring/Grbl_Esp32/wiki/Spindle-Types) I saw a section "relay". Fine, but one thing that looks reluctant to me is that I should not use SPINDLE_TYPE SPINDLE_TYPE_RELAY, but SPINDLE_TYPE SPINDLE_TYPE_PWM instead. Listed differently in the "What is a Spindle" section.

Attached you´ll find my machine config file, maybe someone has a hint for me?

Jens

---
```
#define MACHINE_NAME            "MACHINE_ESP32 XYZA"

#ifdef N_AXIS
        #undef N_AXIS
#endif
#define N_AXIS 4

#define X_STEP_PIN              GPIO_NUM_12  /* labeled Y */
#define X_DIRECTION_PIN         GPIO_NUM_14  /* labeled Y */
#define Y_STEP_PIN              GPIO_NUM_26  /* labeled X */
#define Y_DIRECTION_PIN         GPIO_NUM_15  /* labeled X */
#define Z_STEP_PIN              GPIO_NUM_27  /* labeled Z */
#define Z_DIRECTION_PIN         GPIO_NUM_33  /* labeled Z */
#define A_STEP_PIN              GPIO_NUM_25  //ok
#define A_DIRECTION_PIN         GPIO_NUM_22  //ok

#define SPINDLE_TYPE SPINDLE_TYPE_PWM
#define SPINDLE_OUTPUT_PIN GPIO_NUM_35

// #define SPINDLE_PWM_PIN         GPIO_NUM_35
// #define SPINDLE_ENABLE_PIN      GPIO_NUM_2 //o??

#define LIMIT_MASK              B11
#define X_LIMIT_PIN             GPIO_NUM_17 //ok
#define Y_LIMIT_PIN             GPIO_NUM_4  //ok
#define Z_LIMIT_PIN             GPIO_NUM_16 //ok
#define A_LIMIT_PIN             GPIO_NUM_21 //ok

#define STEPPERS_DISABLE_PIN    GPIO_NUM_13 //ok

// #define COOLANT_MIST_PIN        GPIO_NUM_21
// #define COOLANT_FLOOD_PIN       GPIO_NUM_25

#define PROBE_PIN               GPIO_NUM_32 //ok

//#define CONTROL_SAFETY_DOOR_PIN GPIO_NUM_35  // needs external pullup
#define CONTROL_RESET_PIN       GPIO_NUM_34  // needs external pullup
#define CONTROL_FEED_HOLD_PIN   GPIO_NUM_36  // needs external pullup
#define CONTROL_CYCLE_START_PIN GPIO_NUM_39  // needs external pullup
```

floggy22, 2020-06-22:
Hi,
you can not use Pin 35 as am output. It is input only.  
https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

JensHauser, 2020-06-22:
So simple, I see it now. Thank you so much, I´ll take care of the hardware and then check the software settings.

closed 2020-06-23 by bdring

JensHauser, 2020-06-29:
Back again, now with GIO PIN 32 for the spindle relay on/off.

Two questions:

1.
Whatever I do configure in the current master branch, Arduino serial console says [MSG:No spindle pin defined] and there is no change at the multimeter.

Maybe related to these issues?
https://github.com/bdring/Grbl_Esp32/issues/431
https://github.com/bdring/Grbl_Esp32/issues/439

2.
In the docs I saw a section "relay". Fine, but one thing that looks reluctant to me is that I should not use SPINDLE_TYPE SPINDLE_TYPE_RELAY, but SPINDLE_TYPE SPINDLE_TYPE_PWM instead. Listed differently in the "What is a Spindle" section.