bdring/Grbl_Esp32 #452: What happens when two interfaces are simultaneously run

karoria, 2020-06-23:
Hi to all great developers.

I want to get clarified about what happens when I use SD card embedded web interface and a USB serial interface (say CNCJS) simultaneously. Can I connect to the USB serial port when I am already running a program from SD? If so, will a feedhold on cncjs interface can actually hold the job? I mean both the connection are in sync realtime?

I am asking this as I haven't used SD card feature till date. I always run my machine through usb serial and cncjs-shopfloor-tablet. Due to limited RPi capabilities, big programs (say more than 10MB) don't get loaded on cncjs and the interface hangs as it has graphic viewer also. So, I was thinking to use SD card for bigger programs as its interface doesn't have graphic and I think it is not loading the full program at once, rather it sends line by line (streaming mode). So bigger programs should run nicely in that interface.

KIndly correct me if I am wrong. Please also suggest if there is better way for my problem.

Best,
Ravi Karoria

bdring, 2020-06-23:
There is nothing preventing you from running more than one interface at the same. You could use the serial port and the web to control and monitor a running job. You, as the administrator of the system, are responsible to make sure you do not do stupid things like trying to send gcode from 2 interfaces at the same time.

I often run jobs from SD via serial port and use my phone to check status remotely.

karoria, 2020-06-23:
Thanks a lot for prompt response. Can I run large (say 40MB) programs from SD without any problem?

bdring, 2020-06-23:
Yes.

closed 2020-06-24 by karoria