bdring/Grbl_Esp32 #454: Steppermotor stuttering

KingMo5h, 2020-06-26:
Please answer the following questions.

What version of the firmware are you using?
newest Dev
Is the problem repeatable?
yes
Under what conditions does the bug occur?
If steps/mm are high, like standart 100 steps/mm stepper stutters but then moves. If it is even higher 255 steps/mm stutters and doesnt move

Strange behaviour, if steps/mm are high it stutters and dont move, if they are low (tested with 32) it drives smooth without stuttering.
Is it something software related or hardware?
Iam using the Dev software, I2s and trinamic xyzabc just changed some pins there.

KingMo5h, 2020-06-26:
Nevermind, default settings was wrong. Max amp was on 0.25A, this is way to low.
Btw wasnt there a command or others to activate the extended settings? Couldnt find it anymore.

bdring, 2020-06-26:
Try using $S to access settings if you are using the devt branch.

closed 2020-06-26 by bdring