bdring/Grbl_Esp32 #455: Spindle PWM settings

karoria, 2020-06-26:
Hi friends,
I am using a delta MS300 VFD with 0-10V analog input. I use a pwm to analog converter board for this whhich is readily available in market. I am not getting proper rpm when vfd input voltage is near to zero. So, I have decided to use 3 to 10 V analog to map with 0 to 24000rpm of my 2.2kw spindle. The setting for vfd is already done, but I can't figure out how to set spindle pwm to range from 30% to 100% duty cycle. I want to map 30% duty cycle as 0 rpm and 100% as 24000rpm. I tried $35 =30 and $36 =100 but no luck. Please guide me.

bdring, 2020-06-26:
Try setting your values like this...

```
$30=24000.000
$31=0.000
$32=0
$33=5000.000
$34=0.000
$35=30.000
$36=100.000
```

Then do a reboot by either clicking the reset button on the ESP32 or sending [ESP444]RESTART.

It works for me on my test system.

karoria, 2020-06-27:
Thanks a lot. Will try today only.

karoria, 2020-06-27:
It seems there is a bug. I tried with two type of settings and the results were different. I used DSO to measure duty cycle. Following are settings:

Common settings:
$30=24000.000
$31=0.000
$32=0
$33=5000.000
$34=0.000

Case 1: 
$35=30.000
$36=100.000
Measurements of spindle rpm sent via gcode Vs PWM Duty Cycle
10-12%
1000-14.5%
12000-46.5%
23000-78.4%
24000-99.5%

Case 2: 
$35=60.000
$36=100.000
Measurements of spindle rpm sent via gcode Vs PWM Duty Cycle
10-23.5%
1000-25%
12000-43%
23000-61%
24000-99.5%

My observation from the above readings:
say total blank span out of 100%= 30 in first case and 60 in second case.
name it 3x. So the duty cycle before spindle min rpm shall be  x and after max rpm shall be 2x.
Strange thing is at 23999 rpm the duty cycle remains 80% for first case and at 24000 it becomes 99.5% (theoretically 100%).

I hope this data might be useful and I am able to explain it well. Waiting for prompt response.

bdring, 2020-06-27:
I am not sure why you are having trouble. This is my test with results read from a logic analyzer.

0 = 0%
1 = 30%
1000 = 33%
12000 = 65%
23000 = 97.50%
24000 = 100%

S12000 Example

[image]

The math is quite simple. It uses a [classic map function as described here](https://www.arduino.cc/reference/en/language/functions/math/map/).

Before the map function it applies any speed overrides and makes sure you are in the min/max range.

karoria, 2020-06-27:
I use pin 17 for spindle pwm. I don't use RMT steps. Does it have any connection to this problem? I suspect there is something in firmware to be enabled or disabled before compiling, which I use differently then you.

karoria, 2020-06-28:
OK. I found the Fix. It is those last 11 #define lines of config.h file which are active (not commented out) by default although you are not using piecewise linear spindle. I recommend to comment out those lines by default.
Thanks for time :)

karoria, 2020-06-29:
Reopening. Sorry, I messed up two different folders of code. That was not actual fix. Now I observe that "disabling limit pin pullup" causes the pwm output behave like those readings above. I use inverse logic limit switches and for that I use external pulldown resistors. 
I can't figure out how a limit switch pullup can cause malfunction in spindle pwm output. This happens in latest firmware as well as some 6 months old firmware.

bdring, 2020-06-29:
Can you paste your complete machine definition file here?

karoria, 2020-06-29:
Sure. I am out of station for 2 days. Will paste it when back to work.

karoria, 2020-07-06:
Sorry for long delay in submitting my machine definition file. Actually I was struggling to reproduce the same as I use inverse type settings in my machine PCB for limit switches. Most of the time I check the ESP32 Dev board unattached, I face problems of false triggering hard limits, door alarm etc. Now I am attaching config.h, cpu_map.h and defaults.h files which are as per old firmware. I am also struggling to get new firmware up and running as when I give M3 command, machine alarms for hard limit.
First thing first, I would highly appreciate getting a solution of the problem defined in this thread. I know this might take a good time of your efforts and thank you very much in advance for all that. Waiting for any clues...
[karoria_config.zip](https://github.com/bdring/Grbl_Esp32/files/4878900/karoria_config.zip)

bdring, 2020-07-06:
Could you please also paste your [startup messages](https://github.com/bdring/Grbl_Esp32/wiki/FAQ#submitting-bug-information). I would like to know what version you are running.

karoria, 2020-07-07:
[ESP444]RESTART
[MSG:Restart ongoing]
ok
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Axis count 3]
[MSG:Timed Steps]
[Controller:ENGILABS_V4]
Grbl 1.1f ['$' for help]
[MSG:Check Limits]
[MSG:'$H'|'$X' to unlock]

Where I replaced CPU_MAP_ESP32_V3.5 to ENGILABS_V4 in my file. I don't use any wifi/bluetooth/telnet/sd features and all are disabled in config.h

bdring, 2020-07-07:
That is a very old version of Grbl_ESP32. Many issues have been fixed since then. 

Please use the current version of the master or devt branches.  We can provide support on those versions.

karoria, 2020-07-07:
Ok. I have some issues in latest master too. I will re-try that and update you.