bdring/Grbl_Esp32 #456: Stack smashing protect failure!

Claude54, 2020-06-27:
Please answer the following questions.

What version of the firmware are you using?  Grbl 1.3a

Is the problem repeatable? yes

Under what conditions does the bug occur? 

>  It occurs when i use the fourth axis A. I am sending GCode to bCNC.
> All goes well with 3 axes but it systematically crashes after a while when i use the fourth axis. 

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```

bdring, 2020-06-27:
Could you please post the gcode that causes the problem. I can test it later today. Just a fewest lines needed to repeat the problem.

Have to tried manually sending the gcode via serial monitor? Many gcode senders have trouble with more than 3 axes and it would be helpful to see if that is part of the problem.

Claude54, 2020-06-30:
Sorry for the late reply but I was not available this weekend. I am sending you a piece of GCode which systematically makes GRBL
 crash. I noticed that GRBL crashed when A arrived around -10000.

[test_axe_A.zip](https://github.com/bdring/Grbl_Esp32/files/4853134/test_axe_A.zip)

Claude54, 2020-06-30:
I have tried with other files and always the same, GRBL crashed when A arrived around -10000

bdring, 2020-06-30:
I'll look at it soon. It could be a memory issue.

bdring, 2020-06-30:
I was able to repeat the problem. I think it is due to such a large number not being properly planned for in function...

```
// formats axis values into a string and returns that string in rpt
static void report_util_axis_values(float* axis_value, char* rpt) {
    uint8_t idx;
    char axisVal[10];
    float unit_conv = 1.0; // unit conversion multiplier..default is mm
    rpt[0] = '\0';
    if (report_inches->get())
        unit_conv = 1.0 / MM_PER_INCH;
    for (idx = 0; idx < N_AXIS; idx++) {
        if (report_inches->get())
            sprintf(axisVal, "%4.4f", axis_value[idx] * unit_conv);  // Report inches to 4 decimals
        else
            sprintf(axisVal, "%4.3f", axis_value[idx] * unit_conv);  // Report mm to 3 decimals
        strcat(rpt, axisVal);
        if (idx < (N_AXIS - 1))
            strcat(rpt, ",");
    }
}
```

`char axisVal[10];` does not have wnough room to store it.

This is in report.cpp. You can try changing it to `char axisVal[20];`. It worked for me. I need to look for other areas before updating the code.

Claude54, 2020-06-30:
Yessss..
I have changed the size of axisVal array and it works now.
Thank you very much.

closed 2020-07-01 by bdring