bdring/Grbl_Esp32 #459: Feature Request: Individual Axes Motor Disable

bdring, 2020-06-28:
The Devt branch supports individual signals going to each motor for enable/disable of the motor. Currently if there are multiple signals they all act together like the standard, global disable signal. This was only temporary until there are commands to control them.

Are there any suggestions on how this should work? How do you disable an axis? What happens when an axis is disabled and then you do a move? What happens after that move? Does it create a loss of sync alarm? How does the "step idle delay" come into play?

Please don't suggest Marlin or Reprap commands.

ithinkido, 2020-06-28:
Do stepper motor move to the nearest full step position on disable?

bdring, 2020-06-28:
Currently it goes to whatever microstep is closest to the target. If step idle delay is less than 255, disable will occur that number of microseconds after motion has stopped.

bdring, 2020-06-29:
One method could be to combine **$StepperIdleTime** with a new **$StepperIdleMask** setting. Everything would stay the same except when **$StepperIdleTime** expired, only the axes in the mask would disable. 

If you change **$StepperIdleMask** it would change the current state of the disables, so it could be used to disable an axis on demand. All axes would automatically re-enable on every move like it does currently. 

The advantage is less impact on code and performance stays similar to traditional Grbl.

just-jason, 2020-06-29:
> Currently it goes to whatever microstep is closest to the target. 
>
I do not see how the stepper motor will remain at the nearest *micro-step* position without power. The motor will attempt to allign poles and fall back to the nearest pole pair ( full-step position ) when power is cut. You can feel this motion when turning the shaft by hand. 
In this case there is the possibility for some fairly large errors between desired step count and actual step count considering that most applications will be using micro stepping and could then loose almost half the number of steps as the as the set number of micro-steps.
Although this issue is discussing a software solution, I did also find a possible hardware solution to automatically reduce the holding current when a motor is idle. 

Around page 5
Various Reference Voltage Driving Techniques for Motor ...
https://www.ti.com/lit/sloa170

This should allow the motor to retain its position. 
Expanding on the idea we could look at driving the Vref by software controlled DAC, I am not sure which direction would be best.

ithinkido, 2020-06-29:
> Currently it goes to whatever microstep is closest to the target. 
>
I do not see how the stepper motor will remain at the nearest *micro-step* position without power. The motor will attempt to allign poles and fall back to the nearest pole pair ( full-step position ) when power is cut. You can feel this motion when turning the shaft by hand. 
In this case there is the possibility for some fairly large errors between desired step count and actual step count considering that most applications will be using micro stepping and could then loose almost half the number of steps as the as the set number of micro-steps.
Although this issue is discussing a software solution, I did also find a possible hardware solution to automatically reduce the holding current when a motor is idle. 

Around page 5
Various Reference Voltage Driving Techniques for Motor ...
https://www.ti.com/lit/sloa170

This should allow the motor to retain its position. 
Expanding on the idea we could look at driving the Vref by software controlled DAC, I am not sure which direction would be best.

bdring, 2020-06-29:
@ithinkido Most of us know how motors work.

This is how Grbl currently works. Most people do not set **$StepperIdleTime** to values other than 255. Some machines with lead screws may hold their position. The previous microstep position is resumed once motion starts again.

Some people want to completely disable a motor. This issue is discussing how to do that.

The Motor Class features are a separate topic and can add advanced features, like idle torque control and analog vref control.

just-jason, 2020-06-29:
> @ithinkido Most of us know how motors work.
> 
I have no doubt. That is why is was asking a question and explaining that I could not see something that I guess you could. 
>Some people want to completely disable a motor.
>

MitchBradley, 2020-06-29:
When you completely disable a motor, the position depends on the machine.  On some machines, it might move very little if at all.  On others, for example a ballscrew Z axis, the spindle might fall right down onto the workpiece or table. When a motor is disabled, it is also usually easy for the user to manually reposition, with no way for the software to know.

Given all that, it is prudent to assume that, when a motor has been disabled, the position is no longer reliable.

ithinkido, 2020-06-29:
What are the use cases for this?

bdring, 2020-06-29:
People want to manually move an axis. I have done it myself many times on belt driven machines.

I have also used machines with encoders in manual mode with the axes positions updating.

MitchBradley, 2020-06-29:
Here is a concrete proposal for how the user interface for motor disable could work.  Note that this does not address the details of position management and machine state, it only addresses that command that you use to tell Grbl to initiate the disabling.

New $ command:

$motor/disable=_motor or axis list_

_motor or axis list_ is a case-insensitive sequence of letters, digits, or the comma (,) character, interpreted as follows:
If a letter that names an axis is present, every motor involved with that axis is disabled.
If a digit 0-9 is present, that numbered motor is disabled.
If a comma is present, subsequent fields are comma-delimited, making it possible to use two-digit motor numbers.

Motor numbers are defined for individual boards; there is no implied correspondence between motor numbers and axes.  In most cases, users will disable by axis.  Disabling by motor number is provided for unusual cases.

Here are some examples:

$motor/disable=XZ             - Disables all the motors for the X and Z axes
$motor/disable=X4             - Disables all the motors for the X axis, and also motor number 4
$motor/disable=01,10         - Disables motors 0, 1, and 10
$motor/disable=0,1,10         - Disables motors 0,1, and 10
$motor/disable=x,y              - Disables all the motors for the X and Y axes

MitchBradley, 2020-06-29:
Another use case for motor disable is manual squaring of dual-motor axes.  The various use cases are machine-specific, so it is difficult to generalize.