bdring/Grbl_Esp32 #460: $$ command returns ERROR:3

HuubBuis, 2020-06-28:
Firmware
[MSG:Grbl_ESP32 Ver 1.2a Date 20200613]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]

Is the problem repeatable? Yes

Under what conditions does the bug occur? Sending $$ results in ERROR:3 when using WiFi (ESP3D) or Telnet. When using Bluetooth or serial all works ok.

MitchBradley, 2020-06-28:
The Devt branch seems to be working:

Telnet:
```
wmb@DESKTOP-6I1SH7T:/mnt/c/Users/wmb/Documents/GitHub$ telnet 192.168.2.161 
Trying 192.168.2.161...
Connected to 192.168.2.161.
Escape character is '^]'.

Grbl 1.3a ['$' for help]

Grbl 1.3a ['$' for help]
$$
$$
$0=3
$1=250
etc
```
WebUI:
[image]

HuubBuis, 2020-06-29:
Compiling the Devt branche results in error: sketch\Motors/MotorClass.h:34:68: fatal error: TMCStepper.h: No such file or directory

I did some tests. On the master branch $$ works fine using (WiFi, Telnet, Bluetooth and serial). The problem starts after (my gui) initializes the grbl controller by sending a reset, $X, and reading build info $I. After any of these commands, the $$ command results in error:3  even when using the ESP3D.  A power cycle is needed to get $$ working agian.
I have a lot of trouble using WiFi on this board. The problem could be in the board itself. I have ordered a new board and will continue WiFi testing using this new board.
I closed the issue and will open a new one if the new board has the same problem. Until then, I will use Bluetooth for testing.
Thanks for your support!

closed 2020-06-29 by HuubBuis

MitchBradley, 2020-06-29:
To compile the Devt branch, you must install the TMCStepper library, even if you are not using Trinamic drivers - since Devt includes a runtime configuration mechanism that lets you change the driver type without recompiling.