bdring/Grbl_Esp32 #461: Spindle relay on/off behaviour

JensHauser, 2020-06-29:
Back again from issue [#452](https://github.com/bdring/Grbl_Esp32/issues/451) , now with GPIO_NUM_32 for the spindle relay on/off.

my machine.h
```
#define SPINDLE_TYPE SPINDLE_TYPE_PWM
#define SPINDLE_ENABLE_PIN      GPIO_NUM_32
```

Two questions:

1.
Whatever I do configure in the current master branch, Arduino serial console says [MSG:No spindle pin defined] and there is no change at the multimeter.

Maybe related to these issues?
https://github.com/bdring/Grbl_Esp32/issues/431
https://github.com/bdring/Grbl_Esp32/issues/439

2.
In the docs I saw a section "relay". Fine, but one thing that looks reluctant to me is that I should not use SPINDLE_TYPE SPINDLE_TYPE_RELAY, but SPINDLE_TYPE SPINDLE_TYPE_PWM instead. Listed differently in the "What is a Spindle" section.

JensHauser, 2020-06-29:
A second ago I saw a commit by Bart related to spindle subject. In the [source file](https://github.com/bdring/Grbl_Esp32/blob/master/Grbl_Esp32/Spindles/RelaySpindle.cpp) I saw a comment
```
This is used for a basic on/off spindle All S Values above 0 will turn the spindle on.
```
So, beside the "No spindle pin defined" issue, M3/M5 is not enough, I have to add M3 S1000 or something like that?

JensHauser, 2020-06-29:
Merged the master branch again and voilá: no more error messages.
This here worked for me, probably line 2 and line 3 are doubled.

M3 (without any Sxxx) and M5 commands are fine.

Just the wish to adjust the docs.

```
#define SPINDLE_TYPE SPINDLE_TYPE_RELAY
#define SPINDLE_ENABLE_PIN      GPIO_NUM_32
#define SPINDLE_OUTPUT_PIN GPIO_NUM_32
```

bdring, 2020-06-29:
I do not know why the pin is causing the relay to turn on at boot time. Some pins do change during boot. I list the ones I know about on [this page](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins). You could try a different pin.

We are making major changes to the way pins work. soon it will not require any re-compile to setup or change a machine.

JensHauser, 2020-06-29:
Found a comment in an older ESP32 data sheet.

```
The 32 kHz crystal is internally connected to ESP32’s GPIO32 and GPIO33. To use ADC, Touch or **GPIO functions** of
IO32 and IO33, please remove the 32 kHz crystal and its capacitors — C13 and C17, and solder the 0ohm resistors —
R5 and R6.
```

I asked Espressif, let´s see if they reply.

JensHauser, 2020-07-03:
Espressif wrote:
1. Can you help me catch the output waveform of boot time and press the EN button permanenty of io32? What this waveform looks like.We better analyze the situation for you

2. The descriptions written in the old version are problem, ESP32 DevKitC V4 is not connected 32 kHz crystal 。So this part is not introduced in the new version

So, in general, IO32 and IO33 are ok for being used. I‘ll give them a screenshot of the output curve for further investigation. Then we‘ll know, where to look further

JensHauser, 2020-07-03:
It´s been a wrong dimensioned resistor of the relay. It´s been too small, so even the small current flowing during start of ESP32 triggered the relay.

closed 2020-07-03 by JensHauser