bdring/Grbl_Esp32 #463: CUSTOM PROGRAM FOR Z AXIS FOR PLASMA CUT (THC)

Paulsao, 2020-06-30:
Hello friends
I have the fully functional firmware on the esp32, but it occurred to me to use them on a plasma table for cutting, and I have my TCH for signal inputs. I want to know if it is possible to customize a spindle.h to suit the z axis and receive data in TH32 esp32 and use it.

bdring, 2020-07-06:
That would be a very expensive and time consuming project. I put that on the [long term roadmap](https://github.com/bdring/Grbl_Esp32/wiki/Development-Roadmap#--not-on-near-term-roadmap-now--) in case anyone is looking for a project to work on.

closed 2020-07-06 by bdring

wocpsu, 2020-08-12:
@Paulsao I'm developing torch height control on my fork if you're interested

Paulsao, 2020-08-12:
@wocpsu hi, 
Thanks for answering If I am very interested