bdring/Grbl_Esp32 #465: Can't Get WiFi to Connect

reynolds087, 2020-07-02:
Please answer the following questions.

What version of the firmware are you using?

1.2a

Is the problem repeatable?

Yes

Under what conditions does the bug occur?

I enter the following commands. (Specific information has been omitted for security)

[ESP100]SSIDisHere < ok
[ESP101]PasswordIsHere < ok
[ESP110]STA < ok
[ESP444]RESTART < [MSG:Restart ongoing]
ok

The output is as follows:

ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10864
load:0x40080400,len:6432
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.2a Date 20200514]
[MSG:Compiled with ESP32 SDK:v3.3.1-61-g367c3c09c]
[MSG:Using machine:MPCNC_V1P2]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:PWM spindle Output:2, Enbl:32, Dir:-1, Freq:5000Hz, Res:13bits]
Grbl 1.2a ['$' for help]
[MSG:'$H'|'$X' to unlock]
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]
ok
ok

After the reboot, the ESP32 is back in AP mode, and it never connects to my network. No error messages or anything. The information I am entering works on other devices on my network. The only thing I can think of is maybe the MAC address I am adding to my network's white list isn't the correct MAC? I originally connected to the device in AP mode, and pinged it's default IP. I then took the MAC from my arp table, and added it to the whitelist. Are the AP and STA modes using different MAC addresses, and if so, how do I get the MAC for STA mode? I can't disable MAC filtering on my network to test it without the filter.

luc-github, 2020-07-02:
use `[ESP420]` to know all the mac address
there is also `$I` but it will display only the active one

reynolds087, 2020-07-02:
Thanks, I will test it! Hopefully that fixes it.

reynolds087, 2020-07-02:
I updated the MAC address in my white list with the one associated to STA mode, and it still will not connect. This is the output from [ESP420]

[ESP420] < Chip ID: 54521
CPU Frequency: 240Mhz
CPU Temperature: 75.6C
Free memory: 165.36 KB
SDK: v3.3.1-61-g367c3c09c
Flash Size: 4.00 MB
Available Size for update: 1.87 MB
Available Size for SPIFFS: 169.38 KB
Baud rate: 115200
Sleep mode: None
Web port: 80
Data port: 23
Hostname: grblesp
Current WiFi Mode: AP (24:62:AB:D4:F9:D5)
SSID: GRBL_ESP
Visible: Yes
Authentication: WPA2
Max Connections: 4
DHCP Server: Started
IP: 192.168.0.1
Gateway: 192.168.0.1
Mask: 255.255.255.0
Connected clients: 0
Disabled Mode: STA (24:62:AB:D4:F9:D4)
Notifications: Disabled
FW version: 1.2a (20200514) (ESP32)
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S1000.000]
ok

reynolds087, 2020-07-02:
And this is the ESP3D configuration
[image]

MitchBradley, 2020-07-02:
I suppose you could try getting it to connect to some other access point, as a test.

reynolds087, 2020-07-02:
Thanks, it is working now. I didn't really change anything, just kept restarting and entering the info until it worked. It worked first try with my phone's hotspot though, so it must be something about my access point. Thanks for the help.

closed 2020-07-02 by reynolds087