bdring/Grbl_Esp32 #466: Problem with curves in Lightburn.

mohammadfa91, 2020-07-03:
I'm using lasermode in Lightburn. Both axis work fine in same time like cutting a simple line with slope, but when I decide to cut a circle or a simple spline it stops at the first starting point.

bdring, 2020-07-03:
What stops?

mohammadfa91, 2020-07-04:
Motors stuck in the first point and wont move, until I start to cut another shape.

MitchBradley, 2020-07-04:
I had a situation once where the extent of an arc just barely exceeded a soft limit, causing a hard-to-diagnose stop.  In my case the arc was well-behaved, but I was working ever so slightly too close to the end of the table.
Some CAM software can generate very short arcs that are  ill-behaved under the numerical precision of the controller software.  In that case, the computed arc can sometimes be very large instead of very small, triggering soft limits.
Your problem could well be something different, but this is something to check.

mohammadfa91, 2020-07-04:
all Hardware and Software limits are off. Acceleration is 900 m/sec^2 and max travel is 50000. I have tried acceleration from 100 to 10000 travel speed from 1000 to 100000. The shape is in the middle of working area and I've not limit issue, and there is no error getting back.

mohammadfa91, 2020-07-04:
I've tried with default code and default grbl configs but problem still remains. For cutting curves, stops at first point just 1 second laser signal output.

mohammadfa91, 2020-07-04:
I monitored G-codes from PC to ESP32 the result is:

1. 300mm circle with LaserGRBL (worked)
`G90
M5
G0X300.09Y150.09
M4S255
G3X300.09Y150.09I-150J0F2000
M5`

2. rectangle with LightBurn (worked)
`G00 G17 G40 G21 G54
G90
M4
M8
G0X62Y82
G1Y345S200F6000
G1X482
G1Y82
G1X62
M9
G1S0
M5
G90
G0 X0 Y0
M2`
3. 300m with LightBurn (didn't work)
`G00 G17 G40 G21 G54
G90
M4
M8
G0X57.39Y31.99
G1X54.59Y34.25S200F6000
G1X51.84Y36.58
G1X49.14Y38.97
G1X46.51Y41.42
G1X43.93Y43.93
`

bdring, 2020-07-04:
I am away for the weekend and cannot test. To help debug, can you try without laser mode.

mohammadfa91, 2020-07-04:
I have tested without laser mode ($32=0). problem didn't solve, still not working.

MitchBradley, 2020-07-04:
I don't see a circle in case 3, only G0 and G1 commands. No G2 or G3.

mohammadfa91, 2020-07-04:
Case 3 is the beginning of a circle drawing, after this point ESP stops working like nothing happened. the complete G-Code generated by LightBurn is this:
`G00 G17 G40 G21 G54
G90
M4
M8
G0X57.39Y31.99
G1X54.59Y34.25S200F6000
G1X51.84Y36.58
G1X49.14Y38.97
G1X46.51Y41.42
G1X43.93Y43.93
G1X41.42Y46.51
G1X38.97Y49.14
G1X36.58Y51.84
G1X34.25Y54.59
G1X31.99Y57.39
G1X29.8Y60.25
G1X27.67Y63.17
G1X25.62Y66.13
G1X23.63Y69.15
G1X21.72Y72.22
G1X19.87Y75.34
G1X18.1Y78.5
G1X16.41Y81.71
G1X14.79Y84.97
G1X13.25Y88.27
G1X11.79Y91.61
G1X10.4Y95
G1X9.1Y98.42
G1X7.88Y101.89
G1X6.74Y105.39
G1X5.69Y108.94
G1X4.72Y112.51
G1X3.84Y116.12
G1X3.05Y119.77
G1X2.34Y123.45
G1X1.73Y127.16
G1X1.21Y130.9
G1X0.77Y134.66
G1X0.44Y138.46
G1X0.2Y142.28
G1X0.11Y144.2
G1X0.05Y146.13
G1X0.01Y148.06
G1X0Y150
G1X0.01Y151.94
G1X0.05Y153.87
G1X0.11Y155.8
G1X0.2Y157.72
G1X0.44Y161.54
G1X0.77Y165.34
G1X1.21Y169.1
G1X1.73Y172.84
G1X2.34Y176.55
G1X3.05Y180.23
G1X3.84Y183.88
G1X4.72Y187.49
G1X5.69Y191.06
G1X6.74Y194.61
G1X7.88Y198.11
G1X9.1Y201.58
G1X10.4Y205
G1X11.79Y208.39
G1X13.25Y211.73
G1X14.79Y215.03
G1X16.41Y218.29
G1X18.1Y221.5
G1X19.87Y224.66
G1X21.72Y227.78
G1X23.63Y230.85
G1X25.62Y233.87
G1X27.67Y236.83
G1X29.8Y239.75
G1X31.99Y242.61
G1X34.25Y245.41
G1X36.58Y248.16
G1X38.97Y250.86
G1X41.42Y253.49
G1X43.93Y256.07
G1X46.51Y258.58
G1X49.14Y261.03
G1X51.84Y263.42
G1X54.59Y265.75
G1X57.39Y268.01
G1X60.25Y270.2
G1X63.17Y272.33
G1X66.13Y274.38
G1X69.15Y276.37
G1X72.22Y278.28
G1X75.34Y280.13
G1X78.5Y281.9
G1X81.71Y283.59
G1X84.97Y285.21
G1X88.27Y286.75
G1X91.61Y288.21
G1X95Y289.6
G1X98.42Y290.9
G1X101.89Y292.12
G1X105.39Y293.26
G1X108.94Y294.31
G1X112.51Y295.28
G1X116.12Y296.16
G1X119.77Y296.95
G1X123.45Y297.66
G1X127.16Y298.27
G1X130.9Y298.79
G1X134.66Y299.23
G1X138.46Y299.56
G1X142.28Y299.8
G1X144.2Y299.89
G1X146.13Y299.95
G1X148.06Y299.99
G1X150Y300
G1X151.94Y299.99
G1X153.87Y299.95
G1X155.8Y299.89
G1X157.72Y299.8
G1X161.54Y299.56
G1X165.34Y299.23
G1X169.1Y298.79
G1X172.84Y298.27
G1X176.55Y297.66
G1X180.23Y296.95
G1X183.88Y296.16
G1X187.49Y295.28
G1X191.06Y294.31
G1X194.61Y293.26
G1X198.11Y292.12
G1X201.58Y290.9
G1X205Y289.6
G1X208.39Y288.21
G1X211.73Y286.75
G1X215.03Y285.21
G1X218.29Y283.59
G1X221.5Y281.9
G1X224.66Y280.13
G1X227.78Y278.28
G1X230.85Y276.37
G1X233.87Y274.38
G1X236.83Y272.33
G1X239.75Y270.2
G1X242.61Y268.01
G1X245.41Y265.75
G1X248.16Y263.42
G1X250.86Y261.03
G1X253.49Y258.58
G1X256.07Y256.07
G1X258.58Y253.49
G1X261.03Y250.86
G1X263.42Y248.16
G1X265.75Y245.41
G1X268.01Y242.61
G1X270.2Y239.75
G1X272.33Y236.83
G1X274.38Y233.87
G1X276.37Y230.85
G1X278.28Y227.78
G1X280.13Y224.66
G1X281.9Y221.5
G1X283.59Y218.29
G1X285.21Y215.03
G1X286.75Y211.73
G1X288.21Y208.39
G1X289.6Y205
G1X290.9Y201.58
G1X292.12Y198.11
G1X293.26Y194.61
G1X294.31Y191.06
G1X295.28Y187.49
G1X296.16Y183.88
G1X296.95Y180.23
G1X297.66Y176.55
G1X298.27Y172.84
G1X298.79Y169.1
G1X299.23Y165.34
G1X299.56Y161.54
G1X299.8Y157.72
G1X299.89Y155.8
G1X299.95Y153.87
G1X299.99Y151.94
G1X300Y150
G1X299.99Y148.06
G1X299.95Y146.13
G1X299.89Y144.2
G1X299.8Y142.28
G1X299.56Y138.46
G1X299.23Y134.66
G1X298.79Y130.9
G1X298.27Y127.16
G1X297.66Y123.45
G1X296.95Y119.77
G1X296.16Y116.12
G1X295.28Y112.51
G1X294.31Y108.94
G1X293.26Y105.39
G1X292.12Y101.89
G1X290.9Y98.42
G1X289.6Y95
G1X288.21Y91.61
G1X286.75Y88.27
G1X285.21Y84.97
G1X283.59Y81.71
G1X281.9Y78.5
G1X280.13Y75.34
G1X278.28Y72.22
G1X276.37Y69.15
G1X274.38Y66.13
G1X272.33Y63.17
G1X270.2Y60.25
G1X268.01Y57.39
G1X265.75Y54.59
G1X263.42Y51.84
G1X261.03Y49.14
G1X258.58Y46.51
G1X256.07Y43.93
G1X253.49Y41.42
G1X250.86Y38.97
G1X248.16Y36.58
G1X245.41Y34.25
G1X242.61Y31.99
G1X239.75Y29.8
G1X236.83Y27.67
G1X233.87Y25.62
G1X230.85Y23.63
G1X227.78Y21.72
G1X224.66Y19.87
G1X221.5Y18.1
G1X218.29Y16.41
G1X215.03Y14.79
G1X211.73Y13.25
G1X208.39Y11.79
G1X205Y10.4
G1X201.58Y9.1
G1X198.11Y7.88
G1X194.61Y6.74
G1X191.06Y5.69
G1X187.49Y4.72
G1X183.88Y3.84
G1X180.23Y3.05
G1X176.55Y2.34
G1X172.84Y1.73
G1X169.1Y1.21
G1X165.34Y0.77
G1X161.54Y0.44
G1X157.72Y0.2
G1X155.8Y0.11
G1X153.87Y0.05
G1X151.94Y0.01
G1X150
G1X148.06Y0.01
G1X146.13Y0.05
G1X144.2Y0.11
G1X142.28Y0.2
G1X138.46Y0.44
G1X134.66Y0.77
G1X130.9Y1.21
G1X127.16Y1.73
G1X123.45Y2.34
G1X119.77Y3.05
G1X116.12Y3.84
G1X112.51Y4.72
G1X108.94Y5.69
G1X105.39Y6.74
G1X101.89Y7.88
G1X98.42Y9.1
G1X95Y10.4
G1X91.61Y11.79
G1X88.27Y13.25
G1X84.97Y14.79
G1X81.71Y16.41
G1X78.5Y18.1
G1X75.34Y19.87
G1X72.22Y21.72
G1X69.15Y23.63
G1X66.13Y25.62
G1X63.17Y27.67
G1X60.25Y29.8
G1X57.39Y31.99
M9
G1S0
M5
G90
G0 X0 Y0
M2`

bdring, 2020-07-04:
G2 and g3 are converted to g1

MitchBradley, 2020-07-04:
So as far as Grbl_Esp32 is concerned, it's not stopping on a curve at all.

MitchBradley, 2020-07-04:
Are there any line breaks?  As given, I see some very long lines that will overflow the input buffer.  Perhaps the line breaks are implied, but if so, it would be better to show the lines explicitly, to avoid any possible confusion or misinterpretation.

mohammadfa91, 2020-07-04:
Sure it has line breaks, consider (0x0A) instead of any spaces. 
I had no issue with native grbl (UNO) with lightburn before.

MitchBradley, 2020-07-04:
So the last move that worked was
```G1X43.93Y43.93```
and then it stopped, before executing the
```G1X41.42Y46.51``` ?

Or did it stop somewhere else inside the sequence
```
G00
G17
G40
G21
G54
G90
M4
M8
G0X57.39Y31.99
G1X54.59Y34.25S200F6000
G1X51.84Y36.58
G1X49.14Y38.97
G1X46.51Y41.42
G1X43.93Y43.93
```

mohammadfa91, 2020-07-05:
I monitored with Serial Analyzer when lightburn was running,  the last line I've catched was G1X43.93Y43.93 and process never continued.

MitchBradley, 2020-07-05:
Have you used lightburn's console to enter the GCode manually?  When I tried that, I saw an "error:20" after the M4 command with $32=0, but things worked with $32=1.  With $32=1, I was able to run a lightburn job that contained a large circle.

MitchBradley, 2020-07-05:
We also need to know which Grbl_Esp32 version and branch you are using, and which machine file, plus any edits that you have made to that machine file or config.h or defaults.h

mohammadfa91, 2020-07-05:
I was using the old version that doesn't have machine.h file and also there was no SPINDLE_TYPE option.
I updated to last version and set `SPINDLE_TYPE `to `SPINDLE_TYPE_LASER`. Now problem solved. but there is some problem with `USE_RMT_STEPS` I'll discuss it in another issue

closed 2020-07-05 by mohammadfa91