bdring/Grbl_Esp32 #467: Weird tty behavior under frequent connect-disconnect screnario

tsipa, 2020-07-04:
Please answer the following questions.

What version of the firmware are you using?
0469e7b of Devt, spi_daisy_4axis_xyz

Is the problem repeatable?
Yes

Under what conditions does the bug occur?

raspberry pi, 4.19.66-v7+, esp32 with ch341

I've tried to reproduce it without using cncjs, with `screen /dev/ttyUSB0`, and failed to get clear repro. I've tried to spawn multiple FDs to TTY, but problem is something else. With CNCJS it's relatively easy to reproduce and makes it completely unusable.

open browser 1, do "connect" against /dev/ttyUSB0
open browser 2, do "connect" against /dev/ttyUSB0
open browser 1, do "disconnect" against /dev/ttyUSB0
open browser 2, do "connect" against /dev/ttyUSB0
open browser 1, do "connect" against /dev/ttyUSB0

After few times doing that you'll get noop /dev/ttyUSB0 which will not recover until you hit reset button on ESP32.
You can attach to this /dev/ttyUSB0 but it won't respond on any commands. When hit reset it will spew normal boot message.
No `echo '[ESP444]RESTART' > /dev/ttyUSB0` will work, only physical reset.

tsipa, 2020-07-04:
It doesn't seems like it's a problem with uart driver or board itself, at least with
`esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /root/zalupa/boot_app0.bin 0x1000 /root/zalupa/bootloader_qio_80m.bin 0x10000 /root/zalupa/Grbl_Esp32.ino.bin 0x8000 ~/zalupa/Grbl_Esp32.ino.partitions.bin`
i can reset board

MitchBradley, 2020-07-04:
Try logging all of the CNCjs messages using -vvv on the CNCjs server command line.

tsipa, 2020-07-04:
Works:
```
2020-07-04T16:34:15.996Z - silly controller:Grbl < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]
2020-07-04T16:34:15.998Z - silly controller:Grbl < ok
2020-07-04T16:34:15.998Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P|WCO:0.000,0.000,0.000>
2020-07-04T16:34:16.245Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P|Ov:100,100,100>
2020-07-04T16:34:16.495Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:16.496Z - silly controller:Grbl < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]
2020-07-04T16:34:16.498Z - silly controller:Grbl < ok
2020-07-04T16:34:16.765Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:17.015Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:17.017Z - silly controller:Grbl < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]
2020-07-04T16:34:17.019Z - silly controller:Grbl < ok
2020-07-04T16:34:17.273Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:17.526Z - silly controller:Grbl < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]
2020-07-04T16:34:17.528Z - silly controller:Grbl < ok
2020-07-04T16:34:17.528Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:17.781Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:18.033Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:18.035Z - silly controller:Grbl < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]
2020-07-04T16:34:18.036Z - silly controller:Grbl < ok
2020-07-04T16:34:18.295Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
2020-07-04T16:34:18.547Z - silly controller:Grbl < <Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P|WCO:0.000,0.000,0.000>
2020-07-04T16:34:18.549Z - silly controller:Grbl < [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0.000]
2020-07-04T16:34:18.551Z - silly controller:Grbl < ok
2020-07-04T16:34:18.712Z - debug service:cncengine socket.close("/dev/ttyUSB0"): id=uJ-p7hg2nL1vE3ikAAAC     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18339:25)
```
One more connect-disconnect, still works:
```
2020-07-04T16:34:19.063Z - debug service:cncengine socket.list(): id=d8IvgPu2GTW1Hgo1AAAB     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18241:25)
2020-07-04T16:34:19.079Z - debug service:cncengine socket.list(): id=uJ-p7hg2nL1vE3ikAAAC     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18241:25)
2020-07-04T16:34:19.095Z - debug service:cncengine socket.list(): id=uJ-p7hg2nL1vE3ikAAAC     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18241:25)
2020-07-04T16:34:20.317Z - debug service:cncengine socket.open("/dev/ttyUSB0", {"controllerType":"Grbl","baudrate":115200,"rtscts":false}): id=uJ-p7hg2nL1vE3ikAAAC     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18269:25)
2020-07-04T16:34:20.323Z - debug controller:Grbl Add socket connection: id=uJ-p7hg2nL1vE3ikAAAC     at GrblController.addConnection (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:9277:26)
2020-07-04T16:34:20.345Z - debug controller:Grbl Connected to serial port "/dev/ttyUSB0"     at SerialPort.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:9211:28)
", {"__sender__":"bb3c6791-7509-4c0d-aabb-e08bb3d15a75"}): id=uJ-p7hg2nL1vE3ikAAAC     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18379:25)
2020-07-04T16:34:22.270Z - silly controller:Grbl > $$
2020-07-04T16:34:22.275Z - silly controller:Grbl < $0=3
2020-07-04T16:34:22.279Z - silly controller:Grbl < $1=250
2020-07-04T16:34:22.284Z - silly controller:Grbl < $2=0
2020-07-04T16:34:22.288Z - silly controller:Grbl < $3=0
2020-07-04T16:34:22.290Z - silly controller:Grbl < $4=0
2020-07-04T16:34:22.293Z - silly controller:Grbl < $5=1
2020-07-04T16:34:22.296Z - silly controller:Grbl < $6=0
2020-07-04T16:34:22.299Z - silly controller:Grbl < $10=1
2020-07-04T16:34:22.302Z - silly controller:Grbl < $11=0.010
2020-07-04T16:34:22.304Z - silly controller:Grbl < $12=0.002
2020-07-04T16:34:22.307Z - silly controller:Grbl < $13=0
2020-07-04T16:34:22.310Z - silly controller:Grbl < $20=0
2020-07-04T16:34:22.312Z - silly controller:Grbl < $21=0
2020-07-04T16:34:22.333Z - silly controller:Grbl < $22=0
2020-07-04T16:34:22.336Z - silly controller:Grbl < $23=3
2020-07-04T16:34:22.339Z - silly controller:Grbl < $24=200.000
2020-07-04T16:34:22.341Z - silly controller:Grbl < $25=2000.000
2020-07-04T16:34:22.344Z - silly controller:Grbl < $26=250.000
2020-07-04T16:34:22.347Z - silly controller:Grbl < $27=1.000
2020-07-04T16:34:22.349Z - silly controller:Grbl < $30=1.000
2020-07-04T16:34:22.352Z - silly controller:Grbl < $31=0.000
2020-07-04T16:34:22.355Z - silly controller:Grbl < $32=0
2020-07-04T16:34:22.358Z - silly controller:Grbl < $N1=
2020-07-04T16:34:22.361Z - silly controller:Grbl < $N0=
2020-07-04T16:34:22.363Z - silly controller:Grbl < $100=800.000
2020-07-04T16:34:22.366Z - silly controller:Grbl < $101=800.000
2020-07-04T16:34:22.369Z - silly controller:Grbl < $102=800.000
2020-07-04T16:34:22.372Z - silly controller:Grbl < $110=1000.000
2020-07-04T16:34:22.375Z - silly controller:Grbl < $111=1000.000
2020-07-04T16:34:22.378Z - silly controller:Grbl < $112=500.000
[... cut ...]
", {"__sender__":"bb3c6791-7509-4c0d-aabb-e08bb3d15a75"}): id=uJ-p7hg2nL1vE3ikAAAC     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18379:25)
2020-07-04T16:34:44.804Z - silly controller:Grbl > $L
", {"__sender__":"bb3c6791-7509-4c0d-aabb-e08bb3d15a75"}): id=uJ-p7hg2nL1vE3ikAAAC     at Socket.<anonymous> (/usr/lib/node_modules/cncjs/dist/cncjs/server/index.js:18379:25)
2020-07-04T16:34:47.609Z - silly controller:Grbl > $r
```

I got repro without multiple browser magic, simply connect-disconnect from a single browser window.

I did strace on cncjs process and it's clearly do write("$$")  to /dev/ttyUSB0 FD.

MitchBradley, 2020-07-04:
I have to leave soon for a Fourth of July celebration so I will not be able to work with you in detail for several hours.  Since you seem to have a lot of debugging skills, I will give you some avenues to explore.

1. Others have reported issues with CH34x serial chips; the problems went away with a CP2102-equipped ESP32.  I have had problems with CH340 myself - it would make a Mac computer crash under certain conditions.  I think that Mac problem was a bug in the driver.  I just mention that as support for my general distrust of CH34x.

One way to investigate this further would be to do some kernel-level debugging of the CH341 driver.  I see that you can talk to the driver with esptool, but it is still possible that the driver is getting hung in some way that affects only the way that CNCjs uses it.

2. Another source of problems in CNCjs-land has been the nodejs serialport module.  You might get more clues by adding debugging information in that module.  You could debug either in the Javascript portion of the module in node_modules/@serialport/bindings/lib/*.js or in the system-dependent binary code that is compiled from node_modules/@serialport/bindings/src/serialport_unix.cpp .  I don't have time right now to give complete instructions, but perhaps you already know how to rebuild node modules or can figure it out from the internet.

3. Another approach would be to create a nodejs test program that uses the serialport module in a similar way to cncjs, opening, sending $$, closing, repeat.

4. Finally, you could try with a different GRBL controller if you have one.  If that fails, it points toward CNCjs and its serialport infrastructure as the cause of the problem.  If it succeeds, it doesn't tell us much because CH34x chips are unusual in GRBL systems, so the problem could still be anywhere.

Sorry, have to go now.

tsipa, 2020-07-04:
Ugh, i really apologize, i had another device connected with CH34x uart and confused it with esp32. The problem happen on CP210x. I updated to description.

MitchBradley, 2020-07-04:
One final thought - the other problem report from CH340 ESP32 was consistent with the hypothesis that opening and closing the serial port could be causing glitches on the ESP32 enable line, thus partially resetting the ESP32 and getting it into a strange state.  That hypothesis is consistent with your results.  In particular, the fact that esptool can recover it is explained by the fact the esptool explicitly does a full hardware reset in order to put the ESP32 into bootloader mode.  That reset involves controlling some lines via modem control signals that go through a cross-coupled latch on the ESP32 dev board.

MitchBradley, 2020-07-04:
The debugging ideas around nodejs serialport still apply, even with CP2102.  Happy debugging.  Really must leave now.,