bdring/Grbl_Esp32 #473: Sequence of invert bits

JensHauser, 2020-07-11:
I want to enhance my machine from 3 (XYZ) to 4 axes (XYZA).

For the settings e.g. in $2, $3, $5 i had standard values
`#define LIMIT_MASK B111
`

Now I have to adjust settings for the new "A" axis and don't know, where to add the new column.
Is it left to X or right to Z to calculate the bits?

[image]

BlueOrangeLive, 2020-07-11:
@JensHauser take a look below
"Grbl_Esp32 / Grbl_Esp32 / Machines / 4axis_external_driver.h"
  from line 59 as an example.

JensHauser, 2020-07-11:
Thank you for this hint.

```
#if (N_AXIS == 3)
        #define LIMIT_MASK      B0111
#else
        #define A_LIMIT_PIN     GPIO_NUM_39
        #define LIMIT_MASK      B1111
#endif
```

I do interpret this as "add a new column left to X", so new axes are added like CBAXYZ, right?

MitchBradley, 2020-07-11:
Just FYI - The Devt branch, soon to be promoted to master, has the new Settings mechanism which includes a user-friendly way to configure mask values like $2 and $3.  The old way, using numeric bitmasks, still works, but now you can also just list the axes.  For example, instead of sending `$2=5` to invert X and Z, you can now send `$2=XZ`.  You can also now use text names in place of numbers, for example `$Stepper/StepInvert=xz`.  With this new way, you don't need the table.

You still need to know the bitmask order for the purpose of configuring the limit mask in the code.  The bits in that mask are ordered as  `Bcbazyx`, i.e. the rightmost bit is x and they increase to the left.

The new [Settings page](https://github.com/bdring/Grbl_Esp32/wiki/Settings) describes this, and the bitmask table therein has been extended and clarified for additional axes.

closed 2020-07-11 by JensHauser