bdring/Grbl_Esp32 #474: Using pin 12

atlaste, 2020-07-11:
While attempting to get my grbl build to work with a custom built setup, i kept having trouble with startup and serial. It seems the issue was pin 12 being used. 

Esp32 seems to have reserved this pin, and if you wire it to a voltage level converter on startup, the thing wont boot.

There are hardware solutions for this, but i was wondering: why use it in the first place, if there are enough pins?

bdring, 2020-07-11:
[Everything we know about pins is here](https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins)

It is generally OK to use pin 12 as an output. Do not drive it during startup.

MitchBradley, 2020-07-11:
There a lot of scenarios where ESP32 does not have enough pins.  There are several "special" pins, so if you avoid using all of them, you run out of pins very quickly.  GPIO12 is quite safe to use as an output that drives a high impedance input, which is the case when it is driving a step pin on a Pololu module.  Most of the machine configurations that use GPIO12 as X_STEP_PIN do in fact use it for a Pololu module.

If you are trying to run it through a bidirectional level converter for the purpose of driving an external driver's optocoupler, expect problems. Such level converters are typically intended for interfacing to high impedance inputs on both sides.  If you need drive an optocoupler, you need a unidirectional level converter with a high impedance input (which will not cause problems for GPIO12), and an output that is rated for the current and voltage that you are feeding to the external device.

atlaste, 2020-07-12:
Perhaps my problem is then that im not using pololu drivers. Im attempting to run an xyz cnc driven by dm556t drivers, with limiters (optocoupled), start/stop/reset and a rs485 vfd. For most of that (rs485, stepper drivers), i need 5V logic and therefore logic level converters.

MitchBradley, 2020-07-12:
You did not say what kind of level converters you are using.  They are not all the same.  Furthermore, optocoupled external drivers are not "5V logic" in the same sense as is used in IC logic families.  Rather, they are current-driven devices that can be used with 5V or other voltages given suitable dropping resistors.  As I said, some kinds of level converters are not at all suitable for that.

atlaste, 2020-07-12:
@MitchBradley I apologise, I'm a software engineer and having the hardest time on hardware. I'll explain what I'm attempting to do in detail. 

The LLC's I'm using are these: https://www.sparkfun.com/products/12009 .

What I'm basically doing is wiring the ESP32 to the LLC (3.3V+gnd included) and wire an external 5V power supply (with a 10uF cap) to the LLC (5V + gnd). I should probably note that I have a 36V power supply, and convert that to 5V using a LM2596 converter ( https://www.amazon.nl/gp/product/B07MY2NTFV/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 ).

As for the DM556T's, the datasheet (https://www.omc-stepperonline.com/download/DM556T.pdf) reads "Pulse active at rising edge; 4-5V when PUL-HIGH, 0-0.5V when PUL-LOW. Minimal pulse width of 2.5μs" -- which is the main reason I am using 5V. In the past, I've used TB6600's which just seemed to have issues with 3.3V logic.

For the RS485 interface, I'm using a MAX485 module https://www.itead.cc/max485-module.html . These are also ran through LLC's for two reasons. First, RX signals can apparently exceed 3.3V, and the ESP32 datasheet tells me it shouldn't exceed 3.6V. Second, I tried running it with an Arduino nano on 3.3V, but for some reason it didn't work with the VFD. Once I added an LLC, the problems seemed to disappear.

The limiters have a separate PCB board with 4N35 optocouplers: https://a360.co/2Bhv1vV and are used with these endstops: https://www.domoticland.com/en/buttons-and-switches/endstop-switch/2702-3d-printers-mechanical-limit-switch-module-v12-end-stop-3d-printer-endstop-3809200633837.html . In the past I had a lot of trouble with electrical interference, and figured this should solve that.

I used to run Marlin with an Arduino Due for this CNC, but want to move back to GRBL for a variety of reasons. However, running a VFD with RS485 is proving to be challenging and the GRBL_ESP32 seems to be the only port supporting that more or less.

As I said, I'm a software engineer by profession, not an hardware engineer... which means I probably made a lot of mistakes down the road...

MitchBradley, 2020-07-12:
Those Sparkfun LLCs are well-suited for interfacing between ESP32 and MAX485, but marginal - if they work all - for ESP32 <-> external driver.  The reason is because inputs on both ESP32 and MAX485 are "high impedance", which means that the switching is based entirely on voltage, with virtually no current flowing either into of out of the input pin.  By virtually no current, I mean current in the nanoAmp or even picoAmp range.

In contrast, the external driver inputs are low-impedance, which means that it is necessary for substantial current to flow - usually in the range of 1 to 10 milliamps.  That current is used to light an infrared LED, the light from which turns on a photosensitive transistor.  The fact that a non-trivial amount of current is involved is part of the noise-immunity - because many noise sources are incapable of inducing that much current in the external wiring.  The other noise immunity factor comes from the electrical isolation between the LED and the phototransistor - since there is no common wire on the two sides, noise that is generated in the driver side cannot easily leak over to the computer side via a common ground connection.

The circuit in that LLC does not have a way to supply enough current on the 5V side to drive the optocoupler reliably.  It is intended to switch voltage at low currents, not to switch current.

To drive an optocoupler well from an ESP32 output, you really either a buffer chip in the 74[A]HCT family, a driver chip like ULN2803A, discrete transistors, or logic-gate MOSFETs.  Bart's new board uses 74AHCT595 shift registers that have good characteristics for driving optocouplers - and also for driving the high-impedance inputs on Pololu modules.

On the limit switch side, you are on the right track with the optocoupled input, but there are additional steps that can be taken.  When an optocoupler switches state, it is possible that the edge rate is rather slow, which means that the signal takes a long time (by electronic standards) to make the transition from low to high.  This is made even worse by filtering that you might add to reduce noise.  The problem with a slow edge rate is that the signal is near the logic  transition voltage for long enough that any residual noise might cause it to switch, then switch back, etc.  The usual solution for this is to add hysteresis to the GPIO inputs - so the low-to-high switching threshold is a few tenths of a volt higher than the high-to-low threshold.  The AVR chips used in Arduinos have hysteresis, but ESP32 GPIOs do not, so ESP32 is more sensitive to noise.  Bart's new input modules have optocouplers, RC filtering, and an external chip to add that hysteresis, so they use all the best practices for noise reduction.

Paulsao, 2020-07-12:
@atlaste I had the same problems when I use the "polulo" converters from 3.3v to 5v, the problem is that the chip is bidirectional and when you leave the pins in the air it takes it as a logical one and enters current into esp23 which, on pin 12 There is a problem and the grbl program has not been executed. So I decided to use the 74ls245 chip and it works fine, but I think I need to use a chip from the HCT family because the frequency response of this family is better

MitchBradley, 2020-07-13:
LS (Tpd ~ 10 ns) is 2x faster than HCT (Tpd ~ 20 ns).  HCT is between 10x and 50x faster than Pololu stepper drivers need (A4988 Tsu ~ 200 ns, Tstep ~ 1 us).   HCT is 500x faster than optocouplers  (Tr ~ 10 us).  So I wonder about the "because the frequency response is better" thing.

Faster is not always better.  Too-fast transition rates induce more noise, so lots of circuits intentionally slow down the edges.

atlaste, 2020-07-13:
@MitchBradley Thank you *so* much for your detailed explanation, greatly appreciated.
The board of @bdring looks nice. The moment I can order it somewhere, I will... currently my CNC is not really working because of the issue you describe with the impedence. 

I am interested though... In the data sheet it sais that the llc's can drive 0.22A max. Per pin, the driver requires 10mA typical (16mA max). I would have expected that to leave plenty of room?

As for a limiters, if I read you correctly, you basically say is that I should add a schmitt trigger to the input pins, right?

Regardless, I'm eager to get the board of bart and get this properly solved... my lack of electrical systems has been dragging my cnc project for way too long to be fun...

MitchBradley, 2020-07-13:
The FET can transfer that much current - but the current has to come from somewhere.  In the low direction the current is going into the ESP32 pin so you could in principle hook up to the low side of the optocoupler.  In the high direction the current is coming from a combination of ESP32 at 3.3V and a resistor to 5V, so the high drive at substantial current is only 3.3V, while the 5V drive is through a 10K resistor which is much to large to supply enough current to drive an optocoupler.  So in other words, the level converter might as well just not be there at all.
A schmitt trigger will help the noise immunity of limit switches.  If you are not having problems, don't bother.
The new board from Bart is getting close.  He has done two revs of the design and I have design reviewed it extensively.  We are very happy with it at the moment.  I assume he will go to production any day now.

bdring, 2020-07-13:
On order from PCBWay.

Schematic

https://github.com/bdring/6-Pack_CNC_Controller/blob/master/CNC_IO_Modules/4x_Opto_Input/V1p1/CNC_Module_Opto_Schmitt_V1p1.pdf

atlaste, 2020-07-14:
@MitchBradley Again, thanks for the details. Looking forward on ordering the board from bart, but in the meanwhile (had some background communication and it will probably take well over a month before it's here from what I understood) I would like something that works as well, so I can get started on the software. After all this, I think the best course of action for me would be to mill a simple PCB that has all the necessary components to get this started. 

I've never had trouble with optocouplers without a schmitt trigger tbh. I've had a lot of trouble with false triggers when I didn't use any optocouplers.

Based on your and bdring's extensive comments I've added a simple schematic: https://a360.co/32iIizA . I've avoided using pin12 entirely. I would be very happy if you could take a quick look at it to see if you spot any issues, before I burn my hardware... :-)

MitchBradley, 2020-07-14:
The 7805 will not do what you want.

1. You can't make 5V from 3V3 with a linear regulator. 
2. The 3V3 pin on an ESP32 module is not a good place to pull current because it goes through a tiny on-module linear regulator. It is better to power secondary regulators directly from the main supply.
3. Any regulators that has a common ground between the input and output sides is not "isolated"

If you really want an isolated secondary for the optos, you need an "isolated DC to DC converter", input from the 5V supply, 5V output.

You have already chosen not to take my advice about schmitt triggers so I won't say any more about that.

The probe pin will likely be noisy.  Resume, abort, reset will be noisy too but if their switches are in the same box and close to the ESP32, they might work anyway.

The ULN2803A is fine for driving external steppers.  The ULN2803A I5 input is missing a connection to the ESP32.

MitchBradley, 2020-07-14:
Oh, I see that you have two different "VCC" nets, one round and one flat.  That sort of naming conflict is really bad practice, akin to having two identically-named-but-distinct identifiers in a program.  It is guaranteed to cause confusion - and already has.

atlaste, 2020-07-14:
@MitchBradley You are right, I just didn't know how to fix that at that time. Fixed version is here: https://a360.co/32iIizA .

atlaste, 2020-07-21:
I'll just close this issue. I couldn't ask for a better answer, and have all the information I could ever ask for. Thanks guys!

closed 2020-07-21 by atlaste

MitchBradley, 2020-07-21:
The updated version of the schematics posted above still has problems.  You cannot use a linear regulator to go from 5V to 5V, as linear regulators need a higher voltage on the input side than the output.  The difference is called the "dropout voltage".  A regulator with 2V dropout, such as 7805, needs 7V on the input to regulate to 5V on the output.  Also it appears that the Y stepper signal that is driven by O5 on the ULN2803 is grounded on the I5 input, so the output will never conduct current.

atlaste, 2020-07-23:
@MitchBradley i just milled and tested the board. It works, so for now I'm happy. Eventually ill probably order one of the modular boards and/or mill some other because its so much fun to make these things :) 

One feature i am really missing though is the ability to tie each limit switch to a separate input. That way, if one of them is malfunctioning, grbl can just tell you. 

I've been thinking about how this could be realised. I figured that you basically need a few pin for state changes (xor of all endstops will do) and an spi bus for which endstop is hit.

MitchBradley, 2020-07-23:
I would do the separate-limit-switch thing by creating an expander module for the new board.  It would have either an I2C expander chip or perhaps an STM32F030 microprocessor.