bdring/Grbl_Esp32 #475: Send hello message on connect instead of init()

atlaste, 2020-07-12:
Some clients like CNCJS use the hello message to figure out if the connection is there. 

When testing with GRBL (Arduino Nano), the behavior is always the same: after a connection has been made with the COM port, the welcome message is sent. Close the connection and open again -- and the welcome message is sent again. 

When connecting with GRBL/ESP32 using USB, the hello message is NOT always sent, and the resulting behavior is that the client doesn't work. 

I've checked the code, my theory is that the init() code sends the 'hello' message instead of when a connection is made to the COM port, and it ends up in the send buffer. If the client quickly connects, it is flushed to the client and works -- but that means subsequent calls and reconnects don't work. (Which is exactly the behavior I see).

I refrained from fixing this, because I do not know if this is actually intended behavior because of dropping connections, where you want to resume instead of starting from scratch. Either way, I would guess that sending the hello message on the _first_ connect is always the desired behavior.

-- unplug, quickly connect --

CNCjs 1.9.22 [Grbl]
Connected to COM7 with a baud rate of 115200
VMDPV_1|1_VMDPV
�
[MSG:Grbl_ESP32 Ver 1.2a Date 20200613]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:PWM spindle Output:15, Enbl:2, Dir:-1, Freq:5000Hz, Res:13bits]
VMDPE_1:2:4062:0:4062|4064_VMDPE
Grbl 1.2a ['$' for help]

-- close, open again (any subsequent attempt) --

CNCjs 1.9.22 [Grbl]
Connected to COM7 with a baud rate of 115200
>

--

MitchBradley, 2020-07-12:
I am a developer on both the CNCjs and Grbl_ESP32 projects, so, given enough time, I should be able to give a definitive answer and solution to this one.  I will dig deeper.

atlaste, 2020-07-13:
> I am a developer on both the CNCjs and Grbl_ESP32 projects, so, given enough time, I should be able to give a definitive answer and solution to this one. I will dig deeper.

I can reproduce this every time and have been a professional software developer for many years, so if you need any more details (with or without code locations), just give me a shout.

MitchBradley, 2020-07-15:
Lacking the following information, I cannot duplicate your setup:
* What computer are you running CNCjs on
* How did you install CNCjs?  (electron app, docker, direct npm install, etc)
* On what flavor of ESP32 are you running Grbl_ESP32?
* What kind of USB serial chip does it have - CP2102, CH340, etc

On my setup, I sometimes see inconsistent startup behavior but most of the time it works.  I have tracked the problem down, at least in part, to the following interesting hardware behavior.  First, it is important to note that nearly all ESP modules connect the SoC's EN (active high to enable, low to reset) line and IO0 "download mode" line to the serial chip's DTR and RTS lines, so that DTR high, RTS low will reset the CPU.

When a program is connected to the serial port, the EN signal remains high as long as the connection persists.  But when there is no connected program, I am seeing periodic low-going pulses on EN.  The pattern is one pulse that lasts for 3 us, then a 750 ms delay and another pulse.  The two-pulse pattern repeats every 5 seconds until a connection is opened.  Furthermore, there is also a pulse when a connection is closed, but not when a connection is opened.

This suggests that something in the USB driver or stack is periodically probing with the result that the ESP32 chip is getting doubly-reset every 5 seconds - at least on this Windows 10 system with a CP2102 USB serial chip.

I tried bypassing the 4-port USB hub that was between the computer and the CP2102; no change.

atlaste, 2020-07-15:
> What computer are you running CNCjs on

Windows 10 Pro x64. The pc I'm running it on is a Dell XPS 15, 9550.

> How did you install CNCjs? (electron app, docker, direct npm install, etc)

Windows installer (https://github.com/cncjs/cncjs/releases), desktop app, x64, 1.9.22.

> On what flavor of ESP32 are you running Grbl_ESP32?

ESP32 DevkitC rev 4 WROOM-32
This one to be exact: AZDelivery ESP-32 Dev Kit C V4 WLAN WiFi Development Board .
Data sheet: https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Datenblatt_a3bf98d8-6a53-4d26-8f1c-c61b1c82af39.pdf?76837 .

> What kind of USB serial chip does it have - CP2102, CH340, etc

CP2102 according to the data sheet.

> First, it is important to note that nearly all ESP modules connect the SoC's EN (active high to enable, low to reset) line and IO0 "download mode" line to the serial chip's DTR and RTS lines, so that DTR high, RTS low will reset the CPU.

I don't see a reset happening when reconnecting. I believe this is exactly the issue. When i reconnect, I can do GRBL commands like $$ to get the current settings.

> I tried bypassing the 4-port USB hub that was between the computer and the CP2102; no change.

I'm not using a hub, it's directly connected to the PC.

 > [EN signal] / double reset

Hmm that sounds feasible. I don't have access to a scope, but can put a multimeter between EN and GND to see if it goes low. If it remains high, there's a different problem. Will have to get back to you on that.

MitchBradley, 2020-07-15:
EN only goes low for 3 microseconds per pulse and the pulses are widely spaced relative to their widths, so it is not likely that you will see them with a multimeter.

closed 2020-11-02 by bdring