bdring/Grbl_Esp32 #481: Best way to take an alarm/fault input

karoria, 2020-07-18:
Hi friends,
I am using easy-servo (stepper with encoders) motors for my mill. I want to know the best practice if one of the drive is in fault (due to position following error or overcurrent or any reason), how to take that signal in grbl_esp32 and configure next actions like feedhold/reset/relay operation/stepper disable. I am using leadshine ES2-DA808 drive which gives Open Collector output upon detecting a fault.
I am inclined to "feedhold" upon detecting such fault. But I want to know the best practice if there is nothing to do with grbl_esp32.
Thanks in anticipation.
Ravi

bdring, 2020-07-18:
I think we need to expand the list of `sys_rt_exec_alarm`  alarms.

currently

```C++
#define EXEC_ALARM_HARD_LIMIT           1
#define EXEC_ALARM_SOFT_LIMIT           2
#define EXEC_ALARM_ABORT_CYCLE          3
#define EXEC_ALARM_PROBE_FAIL_INITIAL   4
#define EXEC_ALARM_PROBE_FAIL_CONTACT   5
#define EXEC_ALARM_HOMING_FAIL_RESET    6
#define EXEC_ALARM_HOMING_FAIL_DOOR     7
#define EXEC_ALARM_HOMING_FAIL_PULLOFF  8
#define EXEC_ALARM_HOMING_FAIL_APPROACH 9
#define EXEC_ALARM_SPINDLE_CONTROL      10
```

We would then need to have some input I/O features with possible interrupts to watch that I/O

They could be generic like 
```C++
#define ALARM_INPUT_1     GPIO_NUM_xx
#define ALARM_INPUT_1_TYPE    EXEC_ALARM_MOTOR_FAULT
```

Eventually the same idea could be done without `#defines`

It could quickly get messy if you want a separate input for each driver.

MitchBradley, 2020-07-18:
The alarm code value is used as follows:

* Several code paths check that it is nonzero
* protocol.cpp calls report_alarm_message with the code, and then `grbl_sendf(.."ALARM:%d\r\n", alarm_code);`
* protocol.cpp checks for the specific values EXEC_ALARM_HARD_LIMIT and EXEC_ALARM_SOFT_LIMIT, treating those as "critical events".  Other alarm values are treated as non-critical

Some GCode senders parse the ALARM:n message.  For example, CNCjs handles it as follows:

* The portion after ALARM: is extracted and spaces between : and n are removed.
* The result is converted to a number.
* The numerical value is looked up in a table that contains the number, a brief description, and a full description.  For example, number 7 is "Homing fail"  ..  "Homing fail. Safety door was opened during homing cycle."
* The table contains entries up to number 9.  Number 10, which is EXEC_ALARM_SPINDLE_CONTROL, is not supported in CNCjs.
* If there is a match in the table, CNCjs sends the following message to the UI agent: ALARM: n brief_description.  This is intended for GRBL 1.1 compatibility
* If there is no match in the table, CNCjs sends the entire ALARM:... message that it received, verbatim.  This is intended for GRBL 0.9 compatibility.

GRBL 0.9 sent alarm messages in the form ALARM: brief_description .  If we presume that GCode senders can handle GRBL 0.9 reporting, then we can send any new alarm types as text instead of numbers.  Sending new alarm types as numbers will not be very friendly, unless we can manage to get senders to update their tables.

The possible failure mode would be if a sender, instead of inspecting the form of the ALARM: message and doing the right thing, were to switch on knowledge of the GRBL version.

If we assume that it is acceptable to send alarm messages in GRBL 0.9 text form, here is what we need:

* Each alarm needs a message string and a critical bit.
* The clean way to do that is to create an alarm class with two fields, `bool critical`  and  'const char* message`
* If we want to we can add a third field `int alarmNum` for the old grbl 1.1 compatible number - but I'm not sure that we need it.
* Instantiate that class for every existing alarm, and whenever a new alarm is needed, instantiate it again with the new message.  There should not be a need to assign new numbers, because senders will not know them for new alarms anyway.

The argument to system_set_exec_alarm() changes from uint8_t to Alarm*.

karoria, 2020-07-20:
Thanks for considering it for improvement. For now, can I use step disable pin as input for such alarm to disable all the motors? Or any external arrangement to input to feedhold pin, so that till there is an alarm, user can not resume the program. Or simply operate a relay and cut off the power to drives.

bdring, 2020-07-26:
This has been added to the [roadmap](https://github.com/bdring/Grbl_Esp32/wiki/Development-Roadmap#grbl-note-donations-help-get-roadmap-items-done)...closing

closed 2020-07-26 by bdring