bdring/Grbl_Esp32 #482: Wiki issue regardign BESC setup

BogdanTheGeek, 2020-07-18:
Hi, I cant seem to figure out how to setup my machine to use an ESC for spindle.
which Defines should I have in m machine file?
https://github.com/bdring/Grbl_Esp32/wiki/Spindle-Types#spindle_type_besc
here it says to use SPINDLE_PWM_PIN, but for the rest it is SPINDLE_OUTPUT_PIN
Also, should the machine type be BESC or LASER?

bdring, 2020-07-18:
It should be 

`#define SPINDLE_TYPE       SPINDLE_TYPE_BESC`

Fixed the wiki

closed 2020-07-18 by bdring

BogdanTheGeek, 2020-07-18:
> It should be
> 
> `#define SPINDLE_TYPE SPINDLE_TYPE_BESC`
> 
> Fixed the wiki

https://github.com/bdring/Grbl_Esp32/wiki/BESC-Spindle-Feature#adding-the-feature
this also talks about add_esc_spindle.h
What is that about? is it still relevant?

bdring, 2020-07-18:
Thanks. That page pre-dates the use of standardized spindle classes. It has some good info, so I cleaned it up and cross referenced the pages.