bdring/Grbl_Esp32 #488: $2 step port invert not working?

jimfong1, 2020-07-20:
Please answer the following questions.

What version of the firmware are you using? version downloaded on 7/19/20

Is the problem repeatable? Yes

Under what conditions does the bug occur?  Doesn't invert step pulse, checked with oscilloscope. 

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```

bdring, 2020-07-20:
Please post your boot messages per...

https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages

jimfong1, 2020-07-21:
Sorry,  boot message and config

MSG:Grbl_ESP32 Ver 1.2a Date 20200710]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESPDUINO_32]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:PWM spindle Output:19, Enbl:-1, Dir:18, Freq:5000Hz, Res:13bits]
Grbl 1.2a ['$' for help]
$+
$0=3
$1=255
$2=7
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=1
$24=200.000
$25=2000.000
$26=250
$27=1.000
$30=1000.000
$31=0.000
$32=1
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$80=0
$81=0
$82=0
$83=0
$84=0
$90=0.000
$91=0.000
$92=0.000
$93=0.000
$94=0.000
$100=80.000
$101=80.000
$102=100.000
$110=30000.000
$111=30000.000
$112=1000.000
$120=2000.000
$121=2000.000
$122=200.000
$130=300.000
$131=300.000
$132=300.000
$140=0.250
$141=0.250
$142=0.250
$150=50.000
$151=50.000
$152=50.000
$160=16
$161=16
$162=16
$170=16
$171=16
$172=16
ok

bdring, 2020-07-21:
It appears to work for me. I am using the latest master branch.

```
[MSG:Grbl_ESP32 Ver 1.3a Date 20200720]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESPDUINO_32]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X Axis standard stepper motor Step:GPIO(26) Dir:GPIO(16) Disable:None]
[MSG:Y Axis standard stepper motor Step:GPIO(25) Dir:GPIO(27) Disable:None]
[MSG:Z Axis standard stepper motor Step:GPIO(17) Dir:GPIO(14) Disable:None]
[MSG:Global stepper disable pin:GPIO(12)]
[MSG:No spindle]
more...
```

[image]

jimfong1, 2020-07-21:
I thought I downloaded and compiled the latest version but apparently not. It's working now. Sorry to waste your time. 
Thanks

Regards,

Jim

closed 2020-07-21 by jimfong1