bdring/Grbl_Esp32 #492: GRBL not resetting on [CTRL-X]

atlaste, 2020-07-21:
Please answer the following questions.

1.2a/20200613

Is the problem repeatable?

Yes. If you use Candle (https://github.com/Denvi/Candle) you can reproduce this easily: 

1. Hit a limit switch
2. Click 'reset' button in the 'control' section (top-right screen)
3. Click 'unlock' button in the 'control' section (top-right screen)

Resetting using a hardware switch works fine.

JensHauser, 2020-07-23:
I do have the same issue after a hard limit is touched. Clicking on reset button releases the stepper motor current but that´s it

JensHauser, 2020-07-25:
Reset works with recent fix, alarm stays (hard limit still reached) until homed again. So expected behaviour.

One thing I realized

- The hard limit is touched
- Alarm pops up
- If I click e.g. on Home X or Y or Z, nothing happens (as expected)
- Reseting the user interface induces that these old orders are executed immediately

Idea: better flush the order queue with hitting reset button

bdring, 2020-07-25:
@JensHauser 

I agree, that is the preferred behavior.

bdring, 2020-07-26:
Fixed thanks @MitchBradley !

closed 2020-07-26 by bdring