bdring/Grbl_Esp32 #494: Hard limit

JensHauser, 2020-07-23:
Please answer the following questions.

What version of the firmware are you using?
1.3a.20200720

Is the problem repeatable?
yes

Under what conditions does the bug occur?
- Start Grbl/ESP32
- Home X/Y/Z
- Click "move 1mm towards Y home" several times leads to hard limit. The limit sensor is touched also in reality. Y home is on the other end, see config.
- If I do the same with X or Z, soft limit is regularly touched and alarm pops up.
- Looks like Y soft limit is not taken into consideration

```
$0=3
$1=250
$2=0
$3=2
$4=1
$5=0
$6=1
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=2
$24=200.000
$25=3000.000
$26=250.000
$27=5.000
$30=1000.000
$31=0.000
$32=0
$N1=
$N0=
$100=160.000
$101=160.000
$102=160.000
$110=5000.000
$111=5000.000
$112=1500.000
$120=200.000
$121=200.000
$122=200.000
$130=1000.000
$131=425.000
$132=120.000
```

bdring, 2020-07-24:
It might have to do with the fact that $23 ...  $Homing/DirInvert=2. The Y axis is inverted.

Immediately after you home, send ? to get the position.

Grbl homes into negative space. This is traditional for many CNC machines. Here is my test machine. All max travels are 300mm. This means each axis will have a machine work area of -300mm to 0mm. The homing pull off is 3mm, so you can see that offset in my status report.

```
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
ok
<Idle|MPos:-297.000,-3.000,-3.000,0.000|FS:0,0|Pn:P|WCO:-297.000,0.000,0.000,0.000>
ok
$Homing/DirInvert=X
$Homing/Pulloff=3.000
$X/MaxTravel=300.000
$Y/MaxTravel=300.000
$Z/MaxTravel=300.000

```

Make sure your soft limit test moves will be violating that work area. When using soft and hard limits together make sure you use long moves to test them. If you are sneaking up on the limit, the hard may occur before the soft. If you are 10mm away, try a move of 20 to test the soft limit.

JensHauser, 2020-07-24:
After homing, positions are
`<Idle|MPos:-5.000,-420.000,-5.000|FS:0,0>`

Jogging with 10mm or 100m steps works, GRBL denies with a warning.
Jogging every few seconds with 1mm steps from -420 to -421 to -422 to -423 to -424 to -425 works until at -425 the hard limit comes up.

Maybe I misunderstood s.th., but shouldn't the software deny jogging beyond -420 already, independent of the step length?

bdring, 2020-07-24:
The axis travels are absolute values and are applied accordingly.

We are working on the issue of the reset. I will give a good look at the soft limits again.

JensHauser, 2020-07-24:
Thank you so much

bdring, 2020-07-25:
@MitchBradley found a problem with alarms, including hard limits, and pushed a fix to master.

I will still take a look at soft limits tomorrow.

JensHauser, 2020-07-25:
Maybe helpful in this context, I make use of

```
config.h

#define HOMING_CYCLE_1 (bit(X_AXIS)|bit(Y_AXIS))  // Homes both X-Y in one cycle. NOT COMPATIBLE WITH COREXY!!!
```

bdring, 2020-07-25:
Are you sure you are jogging in machine space? Soft limits use machine space (MPos). Since the alarm does not occur until 5mm (your homing pulloff dist) after from the target, it is suspicious.

Try clearing all your offsets with **$rst=#** and confirm all offsets are removed with **$#** 

You can do a move in machine space with G53. G53 says, for this move I want machine space. So **G53G0X-50** would move to x-50 in machine space. For every move like this you must put G53 first.

Try to move outside your soft limits with **G53G0X-421** (assuming your x MaxTravel is 420) and to **G53G0X1**.

JensHauser, 2020-07-25:
In my case I tried all three approaches based on your guidance:

a) X axis
- G53 G0 X1 // my range is 0 (home) to -1000
- Result is that it throws a soft limit alarm
  <Alarm|MPos:-5.000,-421.000,-5.000|FS:0,0>
- Expected behaviour

b) Y axis
- G53 G0 Y-421 // my range is 0 (home) to -420
- Result is that it physically moves to this position without any alarms/warnings
  <Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
   The same with -422, -423, -424
- At -425 there is an "Alarm" with the alarm icon to its left. No soft or hard limit message, just "Alarm"
- Expected no move

c) Z axis
- G53 G0 Z+1 // my range is 0 (home) to -120
- Result is that it throws a soft limit alarm
  <Alarm|MPos:-5.000,-421.000,-5.000|FS:0,0>
- Expected behaviour

So all in all, Y axis is having an issue.

MitchBradley, 2020-07-25:
In your X case above, Y is already at -421 .  Could that be part of the problem?

JensHauser, 2020-07-25:
Tried again, possible typo, so now full verbose output. Hopefully this gives a hint.

With X axis (expected behaviour)
```
$H
<Home|MPos:-5.000,-420.000,3.537|FS:1500,0|Ov:100,100,100>
<Home|MPos:-5.000,-420.000,1.100|FS:200,0>
<Home|MPos:0.138,-0.138,-5.000|FS:283,0>
ok
<Idle|MPos:-5.000,-420.000,-5.000|FS:0,0>
G53 G0 X1
ALARM:2
Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked, click the Reset Button.
<Alarm|MPos:-5.000,-420.000,-5.000|FS:0,0>
```

With Y axis (no alarm)
```
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
$H
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Home|MPos:-5.000,-421.000,4.475|FS:1500,0|Ov:100,100,100>
<Home|MPos:-5.000,-421.000,-1.800|FS:1500,0>
<Home|MPos:-5.000,-421.000,1.906|FS:200,0>
<Home|MPos:-5.000,-421.000,-2.894|FS:1260,0>
<Home|MPos:-2.006,2.006,-5.000|FS:2142,0>
<Home|MPos:1.969,-1.969,-5.000|FS:283,0>
<Home|MPos:4.819,-4.819,-5.000|FS:0,0|Pn:XY>
ok
<Idle|MPos:-5.000,-420.000,-5.000|FS:0,0>
G53 G0 Y-421
ok
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
```

With Z axis (expected behaviour
```
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
$H
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Home|MPos:-5.000,-421.000,4.475|FS:1500,0|Ov:100,100,100>
<Home|MPos:-5.000,-421.000,-1.800|FS:1500,0>
<Home|MPos:-5.000,-421.000,1.906|FS:200,0>
<Home|MPos:-5.000,-421.000,-2.894|FS:1260,0>
<Home|MPos:-2.006,2.006,-5.000|FS:2142,0>
<Home|MPos:1.969,-1.969,-5.000|FS:283,0>
<Home|MPos:4.819,-4.819,-5.000|FS:0,0|Pn:XY>
ok
<Idle|MPos:-5.000,-420.000,-5.000|FS:0,0>
G53 G0 Y-421
ok
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0>
$H
<Idle|MPos:-5.000,-421.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Home|MPos:-5.000,-421.000,0.081|FS:840,0|Ov:100,100,100>
<Home|MPos:-5.000,-421.000,-3.475|FS:900,0>
<Home|MPos:-5.000,-421.000,2.125|FS:200,0>
<Home|MPos:-5.000,-421.000,4.938|FS:0,0|Pn:Z>
<Home|MPos:5.131,-5.131,-5.000|FS:0,0|Pn:XY>
<Home|MPos:1.163,-1.163,-5.000|FS:283,0>
<Home|MPos:4.569,-4.569,-5.000|FS:283,0>
ok
<Idle|MPos:-5.000,-420.000,-5.000|FS:0,0>
G53 G0 Z+1
<Idle|MPos:-5.000,-420.000,-5.000|FS:0,0>
ALARM:2
Soft limit alarm. G-code motion target exceeds machine travel. Machine position retained. Alarm may be safely unlocked, click the Reset Button.
<Alarm|MPos:-5.000,-420.000,-5.000|FS:0,0|WCO:0.000,0.000,0.000>
<Alarm|MPos:-5.000,-420.000,-5.000|FS:0,0>
```

bdring, 2020-07-25:
What is the response to sending **$MaxTravel**

JensHauser, 2020-07-26:
MaxTravel
```

$X/MaxTravel=1000.000
$Y/MaxTravel=425.000
$Z/MaxTravel=120.000
```

G53 G0 Y+1 leads to an alarm, too

MitchBradley, 2020-07-26:
So the Y limits would be -425 .. 0.  It seems like everything is working correctly.  Y homes to the negative limit which is -425 (since $Homing/DirInvert=Y i.e. $23=2 and $Y/MaxTravel=425), then pulls off by 5 ($Homing/Pulloff i.e. $27 = 5), so Y starts at -420.  The Y limits kick in at -425 and 0.

The pulloff does not affect the limit.

JensHauser, 2020-07-26:
Thanks. Seems I have a misunderstanding here.

The wiki says
`Soft limits use Grbl's internal position values to determine if the the motion will crash into the ends. It uses the max travel ($13x) to determine this. `

I thought soft limits would give me a save corridor for e.g. jogging to both ends of an axis and avoid hard limits and resets. It happens sometimes jogging over the machine limits and a friendly "Jog target exceeds machine travel. Jog command has been ignored" would be nice.

But it seems to be different, so I have to be more careful and hit the reset button in case I exceeded the limits around home.

bdring, 2020-07-26:
It should not crash into the ends. You might trigger the limit switch, but the machine is not trying to go past the limit switch. All good limit switches have some over travel.

If you trigger an alarm by hitting a soft or hard limit, it is no big deal.

I am going to close this, because all issues have been resolved.

closed 2020-07-26 by bdring