bdring/Grbl_Esp32 #495: Tool measurement

JensHauser, 2020-07-24:
Hi,

for some time I´ve been working with a X/Y/Z edge finder like this and it worked

[image]

Now I bought this little thing in order to be able to change a tool during machining a job:
[image]

Seems "G43.1 Dynamic tool length offset" is a subject here. In other forums they use some scripting language to calculate "old" and "new" offset and tell GRBL to use the difference. But with current GRBL and ESP32 web interface? 

It´d be nice to
* home X/Y/Z
* use a the aluminum plate to find G54
* start with tool #1
* move to tool change position
* measure (whatever that means) the new tool
* go on with machining

Or am I on the wrong track?

JensHauser, 2020-07-26:
Is it possible to get to known the 
* current Z position in WCS or
* the location after G38.2

in the
`void user_tool_change(uint8_t new_tool)`
function?

This could help to perform the calculation between old tool and new tool position

bdring, 2020-07-26:
Yes,

Look at how reporting.cpp reports these values.

JensHauser, 2020-07-27:
While learning C and how to [fork](https://github.com/JensHauser/Grbl_Esp32_JH) GRBL/ESP32 in order to have some playground for the tool change functionality, there some questions.

My machine is called 4axis_xyxz (links to [4axis_xyxz.h](https://github.com/JensHauser/Grbl_Esp32_JH/blob/master/Grbl_Esp32/Machines/4axis_xyxz.h) and [4axis_xyxz.cpp](https://github.com/JensHauser/Grbl_Esp32_JH/blob/master/Grbl_Esp32/Custom/4axis_xyxz.cpp))

The custom init and custom tool change functions are triggered and basically working.

I added some G code to move to a tool change position, hold, go and press the Z probe button (quick and then slow).

A few questions

1. The slow Z probe (G38.2 Z-2 F30) doesn't work. Manual execution of that G code works, but not in the tool change function. InputBuffer full?

2. When I call getLastZProbePos() in order to get the position where it just hit the button, I get values from the event before. I´d like to have the current value.

3. Some G code is executed and then I want to stop, to some calculations and proceed. In reality, the whole function seems to run through all function content at once?

My guess is, beyond stupid beginners mistakes, something with time slices of ESP32. If that was true, could you please give some advice to me how to dive deeper?

bdring, 2020-07-27:
I am headed out of town for a week to visit family. I won't be able to review or test any code. I also can't donate much time to non standard machines.

Creating custom machine features in Grbl_ESP32 is powerful, but requires a good understanding of Grbl_ESP32 and C++/RTOS development.

Grbl buffers commands. As each command finished, it will accept a new command. If you have 3 commands and there is room in the buffer, it may appear that Grbl instantly executed 3 commands. It just put them in the queue and will execute them when it can. There are ways to force Grbl to wait in a function until the queued commands are done.

You can also create a new task and state machine. The state machine will execute commands and loop around waiting for a state change. The loop's separate task does not affect the main tasks performance.

===========================================================

I did [this machine](https://www.youtube.com/watch?v=9MAoRSXqEfg) in 8-bit grbl. I plan to convert it to Grbl_ESP32/[6 Pack](https://github.com/bdring/6-Pack_CNC_Controller) with automatic Z touch and tool length offset compensation in late August (after the dust settles from the 6 Pack controller production)

Please don't ask about it until then.

JensHauser, 2020-07-31:
I´d like to make use of the tool change functionality for me, learn C and contribute to the project in reverse sequence :-)

So I spent many hours with Bart´s advices and hope to bring some benefit.

Output is a basically working manual tool change function with Z probe as user machine ([4axis_xyxz.h](https://github.com/JensHauser/Grbl_Esp32_JH/blob/master/Grbl_Esp32/Machines/4axis_xyxz.h) and [4axis_xyxz.cpp](https://github.com/JensHauser/Grbl_Esp32_JH/blob/master/Grbl_Esp32/Custom/4axis_xyxz.cpp)).

How does it work so far?
- Tool change ("Tx M06") is triggered by typing in ESP3D
- Measure tool length (only once for the first tool)
- Go to tool change position. Do the tool change manually and press "resume"
- Measure tool length of new tool
- calculate G43.1 and set it

Technically?
- The state machine is a RTOS task
- The RTOS task is started in "machine_init" and triggered in "user_tool_change" by the initial machine state (an integer value)
- Typing in "T1 M06" is fine, all steps are performed, including G38.2 and G43.1

What´s open?
- There is an issue with the calling function "user_tool_change" while the task is running
- Once the trigger for the task is set, "user_tool_change" returns and goes on with the main g code
- It doesn´t wait for the final state of the task (I know why)

I don't know what do to in the do/while loop unitl the final state is achieved. Sleep/time/etc. didn't work, the machine doesn't take a move then.

So far I have this code, but it needs improvement (status output and machine moves are out of sync):

`void user_tool_change(uint8_t new_tool)
{
	newtoolNo = new_tool;
	tc_state = TOOLCHANGE_INIT; // trigger to let the RTOS task start working.

	while (tc_state != TOOLCHANGE_IDLE)
       {
            sleep(25);
       }

      return;
}`

Does anybody have an idea how to get back from RTOS task with a final state and then(!) return from "user_tool_change"?

JensHauser, 2020-10-02:
The Atari example is impressive, but I have the feeling that a comprehensive knowledge about the grbl programming context is necessary to implement this for me.
Is there any plan to implement a wiki page to give some advice to program the  tool change/ tool length offset feature individually?

JensHauser, 2021-01-10:
Hi @bdring,

some time ago we changed thoughts about tool measurement.
I remember you thought about implementing from your recent projects some months ago:

"I did this machine in 8-bit grbl. I plan to convert it to Grbl_ESP32/6 Pack with automatic Z touch and tool length offset compensation in late August (after the dust settles from the 6 Pack controller production)"

I´m not sure about your plans - this enhancement request couldn´t be found on the roadmap.
The link in the youtube video is not working.

Tried on my own (see fork above), it´s working standalone, but I have an issue with multi-tasking. So some more professional review could be helpful.

Maybe it´s a communication thing and it get lost or I misunderstood your timeline/roadmap, would you mind helping out? Thank you.