bdring/Grbl_Esp32 #503: DC Encoder motor support

irshadmgs011, 2020-08-01:
https://youtu.be/PU6tgQ84CEA 
Esp32 grbl support this motor?

MitchBradley, 2020-08-01:
Grbl_esp32 supports RC servo motors but there is very little detailed information in that video so it is hard to say exactly what would be necessary to hook up that one.

irshadmgs011, 2020-08-02:
https://youtu.be/aNqq-6n_vqs
Cricut machine also use this motor

#Aliexpress Rs. 356.40  23%OFF | Mabuchi RS-365PW 16120 Speed Measuring Motor Encoder 6 Pin Plug High Precision Code Tray Optocoupler Motor DC 6V-12V 9V 6700RPM
https://a.aliexpress.com/_dTSIBg4

irshadmgs011, 2020-08-05:
@MitchBradley https://github.com/misan/dcservo

terjeio, 2020-08-05:
@irshadmgs011 The https://github.com/ldocull/MaslowDue port is the only I know of that supports DC motors with encoders and closed loop position control.

MarcinanBarbarzynca, 2020-08-09:
Hey. I'm maslow shield creator. I'm working on something to control normal flat cnc.
I'm trying to pirate your shield and use it as "RC controller for big flat cnc :P witch external H drivers" 

I'm using DC motor on 1 side of TR screw and quadrature encoder on other side of that shaft. LPD3806 600 as encoder.
[image]

Is it capable of controlling 3 axis (3 dc windshield motors) with quadrature encoders?

Maybe i will emulate steppers with that.

MarcinanBarbarzynca, 2020-08-17:
Ok. I've got solution. I will use DCServo https://github.com/misan/dcservo
I need to understand how homing pin works on that arduino shield, than i will adapt that esp32 code to work.

bdring, 2020-08-17:
Is it possible to make that as a CNC I/O Module. If it just requires step/direction/enable and possibly one other I/O, it would not require many changes to Grbl_ESP32.

https://github.com/bdring/6-Pack_CNC_Controller/wiki/CNC-Module-Definition

https://github.com/bdring/6-Pack_CNC_Controller/wiki/CNC-I-O-Module-List

MarcinanBarbarzynca, 2020-08-17:
Hmmm, That's good idea becouse I can create that CNC-I-O Module witch ATTINY85 and get stepper emulated w/o any problem :P 
Whats the price of main board with esp32? Shipping to Poland. 
I've already model my ESP32 module and im prepared pinout for my custom 
https://github.com/MarcinanBarbarzynca/Grbl_Esp32/blob/master/Grbl_Esp32/Machines/DCServo_CNC.h
I'm green at the moment, but I will spend next 6 days tryin to create some procedures for my type of CNC. 
Procedures like:
Handling "I cant keep up" from stepper controller message 
Calibration procedure that compute max feedrate available on mashine. 
Give me 1 more night and I will try to propose commit to discuss.