bdring/Grbl_Esp32 #516: Limitless motions

vanniaz, 2020-08-07:
Hello, I am planning of using Grbl_Esp32 to control 6 stepper motors for a dosing machine. I would use relative distance movements, where distance corresponds to the dose.

Grbl seems like a good choice because it is capable of handling several axes in parallel, but there are a couple of things that worry me:
- the motors will always move in the same direction, could this cause problems with the coordinate system, like overflows in the counters?
- there is no need to handle limit switches or other additional functions which are normally found in a CNC machine, how difficult is it to disable all checks in Grbl?

Of course I could try it, but before embarking in the project I would like to get some advice.

Thank you very much
Giovanni

bdring, 2020-08-07:
The position is stored as a signed 32bit value of motor steps. Do the math to see if you might overflow. With a G28.3, you can reset the machine zero at any time.

Things like limits switches are define in machine definition files. If you do not define limit switches, Grbl_ESP32 ignores all related functions.

vanniaz, 2020-08-07:
Thank you for your reply.
The machine works intermittently so it is perfectly possible to reset the position after each cycle.
I think that this definitely clears my doubts!

closed 2020-08-07 by vanniaz