bdring/Grbl_Esp32 #523: Adding 4th Axis To Already Ganged Axis Machine

jaysettle, 2020-08-08:
Getting more interested in add a 4th axis like this:
https://www.amazon.com/Mophorn-Rotational-2-Phase-Stepper-Engraving/dp/B07VMMWB3K/ref=sr_1_12?dchild=1&keywords=cnc+rotary+axis&qid=1596916480&sr=8-12

I'm currently running **CPU_MAP_LOWRIDER** at version **VER:1.1f.20190708:CUSTOM**. I'm trying to realize what changes I would need to make in firmware but I might be over thinking it. 

- I have a ganged Y axis and the homing cycle to keep the gantry aligned.
- I could use my unused spindle pins as stepper direction and pulse. 

Any help is appreciated.

And while we're on the topic, if I added a 5th axis like this:
https://www.amazon.com/Router-Machine-Rotational-2-Phase-Stepper/dp/B07G94C6DX/ref=sr_1_9?dchild=1&keywords=cnc+rotary+axis&qid=1596916480&sr=8-9
What would a firmware roadmap look like?

Thanks guys.

MitchBradley, 2020-08-08:
We have been working flat-out on a board and supporting firmware for a many-axis machine - https://github.com/bdring/6-Pack_CNC_Controller .  I think everyone's time is better spent making that work as well as possible, instead of trying to shoehorn extra axes into a board that was optimized for a different setup.

bdring, 2020-08-18:
I am about ready to start selling a few to 6 Packs power users. This would be people who have built machines and have used Grbl_ESP32. The hardware is pretty solid and the current state of the firmware is very close done. Final tweaking of the firmware is in process. 

If you are interested in some powerful but "beta" hardware let me know. I have about 15 controllers and a decent stock of most of the modules.

jaysettle, 2020-08-18:
Appreciate the thought. I’m considering it however life is probably the busiest it could get for me now. What kind of commitment or feedback would you be looking for? 

Would it be worth it for me to test the firmware only?

-Jay

bdring, 2020-08-18:
The only firmware changes are tuning the I2S support. That requires hardware to test. The CNC I/O modules run standard GrbL_ESP32 and run great.

I don't need a lot of feedback, but I don't want to sell them to people who are not ready to use them or have no experience putting together a machine.

jaysettle, 2020-08-19:
I could give it a try but I really don't know when I'd get around to installing your new board. Free time pops up ever so often and I'll jump on it then, I'd say within a month. 

Some disclaimers:
- I'm using DB542T's stepper drivers 
- I still want to move to a 4 or 5 axis system, I currently have xyz
- I might actually want my spindle connected to my controller, currently it's not and I don't care
- I'm looking into lasers and mounting my plasma gun on my machine somehow so maybe I could use the extra I/O

Per Mitch's comment "_I think everyone's time is better spent making that work as well as possible, instead of trying to shoehorn extra axes into a board that was optimized for a different setup._" What setup was this board made for?

MitchBradley, 2020-08-19:
Previous controllers had to be optimized for a particular setup because ESP32 pin limitations did not permit a general purpose board.  Each board had to choose the pin assignment carefully, trading off between axes, enables, limits, and miscellaneous control signals for things like spindles, SPI control of smart drivers, alarm switches, etc.

The new board uses I2S expansion to provide a large number of extra signals that can be pulsed at a high rate with excellent pulse rate stability.  That makes it possible to support many motors - up to 9 with per-motor enables or up to 14 without enables.  It frees up the ESP32 pins that would otherwise be used for motors for use as inputs and other purposes.

The board has a modular design so you can plug in adapter modules to customize for specific machine needs.  Quite a few adapters have already been made, including "bulletproof" optocoupled input with excellent signal conditioning, various kinds of spindle control like relays, 5V for lasers, 0-10V for VFDs, and RS485 for VFDs.  Others, including smart modules for interfacing to LCDs and jog pendants, are planned.  The module interface is documented so you can make your own if you need something special.

So, the new board is optimized for flexibility and support of complex machine configurations.

jaysettle, 2020-08-19:
Sounds great. I didn't know it was geared towards being that progressive to not only fit the common opto, spindle and laser needs but include LCD's and pendants and even custom. 

Yeah I'm interested. What's the next step?

bdring, 2020-08-19:
@jaysettle  [Contact me on Slack](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA) via DM.

hpsiepsi, 2020-09-18:
Bdring, if you are still looking for beta testers, I’m interested. I have a home-built machine using esp32 with a hand-wired interface board and external stepper & servo drivers. I had been considering scrapping it for a different control system, but this would be a good option. I’m also a professional working on R&D of laser CNC systems (mostly the laser end)