bdring/Grbl_Esp32 #531: Probe accuracy and i2s stepping

jschoch, 2020-08-10:
Please answer the following questions.

What version of the firmware are you using?

Master and an older version to compare against

Is the problem repeatable?

yes

Under what conditions does the bug occur?

My mill is running VER:1.1F20200319 and I mount a kinematic 3 axis touch probe into the spindle.  I probe off the vise fixed jaw and see the following results:

```
commands run
G38.3F50Z-10
$j=g91 z2 F100
```
kinematic 
##Probing Z
```
[PRB:-42.761,-43.663,-119.478:1]
[PRB:-42.761,-43.663,-119.479:1]
[PRB:-42.761,-43.663,-119.480:1]
[PRB:-42.761,-43.663,-119.480:1]
[PRB:-42.761,-43.663,-119.480:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.482:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.480:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.480:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.480:1]
[PRB:-42.761,-43.663,-119.481:1]
[PRB:-42.761,-43.663,-119.481:1]

```

a 0.002mm per tick dial test indicator was observed and the variability of the stop position was noted and matched the readings.

My bench test setup is running TMC2209 stepper on the Y axis with a small 0.9 deg nema 17 stepper.  The motion is constrained by a MGN12 linear rail and 5mm lead ball screw.  My config, diff vs master, and i2s machine setup can be found here:  https://gist.github.com/jschoch/90c695ced0146bb0ccefcd825b24458a

The dial indicator moved +/- 0.03mm in this setup and the results were:

## Probing Y
```
[PRB:0.000,27.417,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.419,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.420,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.412,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.415,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.425,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.430,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.429,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.435,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.430,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.429,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.429,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.422,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.435,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.435,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.438,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.442,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.443,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.444,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.443,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.440,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.444,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.446,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.448,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.451,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.455,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.451,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.453,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.458,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.462,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.463,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.465,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.467,0.000,0.000,0.000,0.000:1]
[PRB:0.000,27.473,0.000,0.000,0.000,0.000:1]

```

Strangely it seems the readings are ticking up ~0.003 per test run.  The dial indicator did not show this steady increase.

odaki, 2020-08-11:
Thanks for the report.
There did not appear to be any major changes in the flow of Probe's processing from the March 19 edition to the present.

VER:1.1F20200319:
https://github.com/bdring/Grbl_Esp32/blob/d4121b82ff33c67c98c85cc7eabe7ce303a6d31c/Grbl_Esp32/motion_control.cpp#L332

Latest master:
https://github.com/bdring/Grbl_Esp32/blob/35cb11d7231e7729e730a3a64d86ad02768ebdeb/Grbl_Esp32/motion_control.cpp#L341

So my guess is that in probing with I2S, the stepper function may be chopping a little more steps (e.g. one step) than you would expect.
However, we don't know at this time why the value is not the same every time.
While the I2S static mode has little or no delay compared to stream mode, it actually has a delay of 4μs to 8μs. If the Probe is detecting signals during this small delay, then it may be that the Probe point is perceiving the motor position to be slightly further off in the direction of travel than it actually is.

I've never used the Probing feature.
I'm going to check out what's going on at Probing this weekend.
I'm going to look into it to get a more accurate step action.

odaki, 2020-08-12:
I've noticed something.
In I2S static mode, I do not perform an I2S reset from st_reset() or mc_reset().
A change to perform I2S reset even in I2S static mode may be useful to deter slight unintended motor movement.
Resetting I2S immediately after the Probe signal is detected will prevent unintended stepping after the Probe signal is detected and may improve the phenomenon of the motor position moving ahead a little.

jschoch, 2020-08-13:
i switched to devt and here is my results.  it still keeps creeping up.  I don't think I have a mechanical issue.  there is no play when the probe is touching and I wiggle the ball screw.

```
[PRB:0.000,52.383,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.387,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.392,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.392,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.392,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.393,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.395,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.400,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.400,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.400,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.401,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.403,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.405,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.406,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.410,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.414,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.416,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.418,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.423,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.429,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.428,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.433,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.427,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.435,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.430,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.431,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.426,-10.000,0.000,0.000,0.000:1]
```

jschoch, 2020-08-13:
this time i seemed to go the other way.   the dial test indicator had an error range of +/- ~0.01 but the last test showed the same reading on the indicator as the first test in this series did.  the indicator is 0.002 per tick.

```
[PRB:0.000,52.383,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.387,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.388,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.389,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.390,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.391,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.392,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.392,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.392,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.393,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.395,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.400,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.400,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.400,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.401,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.403,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.405,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.406,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.410,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.414,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.416,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.418,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.423,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.429,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.428,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.433,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.427,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.435,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.430,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.431,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.426,-10.000,0.000,0.000,0.000:1]
[GC:G38.3 G54 G17 G21 G90 G94 M5 M9 T0 F50 S0.000]
ALARM:3
[MSG:X  Axis limit switch on pin GPIO(36)]
[... 88 lines cut ...]
[PRB:0.000,52.424,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.427,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.427,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.425,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.424,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.425,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.424,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.425,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.422,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.423,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.422,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.422,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.419,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.418,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.425,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.419,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.418,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.417,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.419,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.420,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.419,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.417,-10.000,0.000,0.000,0.000:1]
[PRB:0.000,52.418,-10.000,0.000,0.000,0.000:1]

```

bdring, 2020-08-13:
I ran a test in the lastest devt branch and static mode. Here are the results.

```
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.584,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.584,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.584,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.584,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.584,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.589,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.589,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.585,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.589,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.587,0.000,0.000,0.000,0.000,0.000:1]
[PRB:-12.586,0.000,0.000,0.000,0.000,0.000:1]
```

[image]

Here is an image of my test setup.  The resolution was 800 step/mm. The speed was 50mm/min. My gcode looks like this.

```
g0x0
g38.2X-3F50
g0x0
g38.2X-3F50
g0x0
g38.2X-3F50
g0x0
g38.2X-3F50
g0x0
g38.2X-3F50
.....many more
```

odaki, 2020-08-15:
Since the phenomenon does not reproduce, it is not clear if it will improve, but we have made a fix for the proper use of I2S modes.
https://github.com/bdring/Grbl_Esp32/pull/547

jschoch, 2020-08-15:
Thanks, i'll close this for now and can re-open if someone else has issues.

closed 2020-08-15 by jschoch