bdring/Grbl_Esp32 #539: Need to map IO to M5stack abd use lcd.

kotopuz, 2020-08-12:
Dear developers. I have m5stack. Do ér have any working example for that with lcd and three buttons used for control?

bdring, 2020-08-12:
We currently do not support the M5 stack or LCDs. We use smartphone, tablets and PCs for display via wifi or bluetooth.

kotopuz, 2020-08-12:
Great. But can you make a sample for m5stack IO mapping please.

bdring, 2020-08-12:
Map to what for what.

Not interested right now. Read the wiki.

closed 2020-08-12 by bdring

f4grx, 2020-08-13:
m5stack has so many hardware bugs and lack of documentation I would not trust it to drive a cnc

moreover many pins are default mapped to the internal hardware (sometimes very strangely) and very few io pins would be available for cnc use.

I definitely do not recommend using an m5stack to drive a cnc in real time. Maybe as DRO by sending commands to a master esp32... but that's a separate project.