bdring/Grbl_Esp32 #546: esp32 commands as wiki page

JensHauser, 2020-08-14:
While setting up a new ESP32 devkit board with a compiled Grbl_Esp32, usage of the direct commands in [docs/command.txt](https://github.com/bdring/Grbl_Esp32/blob/master/doc/Commands.txt) is quite useful. I discovered them by chance and would like to recommend copying them into a wiki page.

If I can contribute, let me know.

MitchBradley, 2020-08-14:
If you type $commands/list or its short form $cmd (case-insenstive) Grbl_Esp32 will list all of the commands including those WebUI ones, giving the preferred new descriptive name, the old cryptic name, and the arguments if any.  Please do create a wiki page, especially if you will sign up to monitor PRs for new commands and update the page.  Commands are added infrequently.  It would be better to de-emphasize, or omit entirely, the ESPxxx names, because they are hard to remember and new commands probably will not have an ESPxxx form.

MitchBradley, 2020-08-14:
A hint for interpreting the arguments as listed by $Commands/List:  An argument is required if there is an "=something" in the list.  If the "something" is in UPPER_CASE, it must be one of the literal values listed.  For example "=FORMAT" means that the argument must be exactly "FORMAT", i.e. $LocalFs/Format=FORMAT.  Another example "=STA|AP|BT|OFF" means choose one of the strings "STA" or "AP" or "BT" or "OFF".  If the "something" is lower case, then the value is arbitrary, although some limitations may apply.

closed 2020-08-14 by MitchBradley