bdring/Grbl_Esp32 #550: sensorless endstops

tmbdev, 2020-08-16:
Could you please document how to enable sensorless endstops for homing; it's unclear to me from the documentation or the discussion in the bug tracker.

(Generally, it would be nice to have an GRBL-Esp32 wiki page with all the configuration parameters and explanations.)

MitchBradley, 2020-08-16:
> (Generally, it would be nice to have an GRBL-Esp32 wiki page with all the configuration parameters and explanations.)

The file config.h contains the parameters and each one has descriptive text.  Thanks for volunteering to create a wiki page with a nicely-formatted version of that information!

bdring, 2020-08-16:
https://github.com/bdring/Grbl_Esp32/wiki/StallGuard-Setup-and-Tuning

bdring, 2020-08-16:
If the wiki is not sufficient, please let us know.

closed 2020-08-16 by bdring