bdring/Grbl_Esp32 #551: Problems Compiling Firmware

water-rhatt, 2020-08-16:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?
YES
What version of the Arduino IDE are you using? 
1.8.13
What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 
1.0.4 (Oct 2, 2019)
Are you using the master branch of Grbl_ESP32?
YES
Have you made any edits or configuration changes (list them) to the  firmware?
NO
Please paste the compiler error text here:
```
In file included from sketch/grbl.h:61:0,
                 from /home/randolph/snap/arduino/current/Arduino/libraries/Grbl_Esp32/Grbl_Esp32.ino:21:
sketch/Motors/MotorClass.h:34:68: fatal error: TMCStepper.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.
```

bdring, 2020-08-16:
That is part of this step.

https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware#copy-libraries

water-rhatt, 2020-08-16:
Thank you.  I didn't notice that there was a second step to add the TMC library.

closed 2020-08-16 by water-rhatt