bdring/Grbl_Esp32 #554: Problems Compiling Firmware in VSCode with platformio

lienbacher, 2020-08-17:
Hello!
It's been a while I've updated grbl_esp32, so I just went and did a fresh git clone, added it as a folder in a fresh workspace and hit build in the bottom toolbar. 

I get a bunch of these errors: 
```
Compiling .pio/build/release/src/print.cpp.o
Assembler messages:
Fatal error: can't create .pio/build/release/src/grbl_sd.cpp.o: No such file or directory
*** [.pio/build/release/src/grbl_sd.cpp.o] Error 1
```

if I hit compile again it will fail at a different point. 
Ultimately if it finally manages to go through everything i get this:

```
Indexing .pio/build/release/libFrameworkArduino.a
Linking .pio/build/release/firmware.elf
.pio/build/release/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x4): undefined reference to `setup()'
.pio/build/release/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `loop()'
.pio/build/release/libFrameworkArduino.a(main.cpp.o): In function `loopTask(void*)':
/Users/wolfgang/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14: undefined reference to `setup()'
/Users/wolfgang/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:17: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
*** [.pio/build/release/firmware.elf] Error 1
```

a google search does not yield any result and I am beyon clueless. Been using platformio for quite a while now and never ran into that kind of issue.

Is my approach to compiling wrong?

HuubBuis, 2020-08-17:
I did the same today and it worked for me!
You have to open the code as a PIO project (file platformio.ini). When building, it can take some time because the ESP32 environment has to be dowloaded and installed first.

lienbacher, 2020-08-17:
that file is detected automatically, and opening a project is more or less the same as having the project folder added to the workspace.

MitchBradley, 2020-08-17:
Try using the Devt branch.

lienbacher, 2020-08-17:
same. 

I have a clue now, just had a popup in the bottom i am using the native python that ships with mac os and I should select a different one. I may have messed up my python setup since I never really actively use it. Python on mac os is a complete mess and I guess it is time to clean it up :(

lienbacher, 2020-08-17:
apart from the python clue, whats really weird is that no error is consistent. Every time I compile the error is slightly different. The only common thing is it always fails (i think it even succeeded once or twice) o.O

MitchBradley, 2020-08-17:
You need Python 3.something.  Python 2 sort of works and sort of does not - you will get a lot of spurious recompiles and other issues with Python 2.

lienbacher, 2020-08-17:
Oh I have python3 installed, I think at least twice. I've probably done everything wrong when installing python on mac os in a few attempts to fix previous attempts of doing it wrong, one of them was just recently, and now it's biting me 🗡️ . explains why my own projects work, they don't use python :D

HuubBuis, 2020-08-17:
I use Python 3.8.3. 32 bit and installed as a single application, not together with programs that depend on it. I run W10

lienbacher, 2020-08-17:
it's not this simple on macOS unfortunately, mainly because it already ships with 2.8 and it cannot be removed. [This](https://opensource.com/article/19/5/python-3-default-mac) outlines the situation a bit.

lienbacher, 2020-08-17:
still busy fixing python, but i'm on a track now. 

One thing I noticed, in my workspace i have a file `Grbl_Esp32.ino.cpp` popping up and disappearing constantly, it's only there for a very brief moment, and then goes away. I don't know if this is related, but if something is modifying the working directory all the time this may be another clue ... This whole issue is very confusing!

lienbacher, 2020-08-17:
not everytime, but almost everytime i hit build I get a different error message 
```
Building in release mode
Compiling .pio/build/release/src/Grbl_Esp32.ino.cpp.o
xtensa-esp32-elf-g++: error: Grbl_Esp32/Grbl_Esp32.ino.cpp: No such file or directory
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pio/build/release/src/src/CoolantControl.cpp.o
Compiling .pio/build/release/src/src/CustomCode.cpp.o
Compiling .pio/build/release/src/src/Eeprom.cpp.o
Compiling .pio/build/release/src/src/GCode.cpp.o
Compiling .pio/build/release/src/src/Grbl.cpp.o
Compiling .pio/build/release/src/src/I2SOut.cpp.o
*** [.pio/build/release/src/Grbl_Esp32.ino.cpp.o] Error 1
Compiling .pio/build/release/src/src/Jog.cpp.o
================================================================================================================================================== [FAILED] Took 3.42 seconds ==================================================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
release        FAILED    00:00:03.424
debug          IGNORED
============================================================================================================================================= 1 failed, 0 succeeded in 00:00:03.424 =============================================================================================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.
```

atlaste, 2020-08-18:
`Grbl_Esp32.ino.cpp` is basically the file that gets generated from the ino file. It is then compiled to an object file and deleted.

Either ways, I would perhaps start with a bare ino file, with an empty setup and loop method and no includes, and try to compile that first.

lienbacher, 2020-08-18:
@atlaste It makes sense to compile it at compile time. However it pops on and off all the time. Not only if I compile. 
I have several esp32 projects large and small that I work on a regular basis that all compile fine. I am using platformio successfully, and this behavior is entirely new to me.

MitchBradley, 2020-08-18:
One thing to try is to turn off every VScode extension that you can find - or at least disable them for this project. Some extensions seem to routinely scan the tree looking for interesting things to index or syntax-check or whatever. That can interfere with compilation it the extension has a file open and it is not accessible to the compilation process during that time.  When I turned off extensions, it stabilized my compilation.  You can then turn on important ones one at a time to see if it starts to break again.

Unfortunately this can be time consuming since some extensions seem to be meta-packages that bring in a whole suite of other ones.

The fact that the .ino.cpp file keeps appearing when you are not explicitly trying to compile suggests that some extension is periodically triggering a compile without your input.

lienbacher, 2020-08-29:
Good input @MitchBradley! Unfortunately even with all extensions except C/C++ and PlatformIO deactivated teh behavior is the same. 
The only thing that is constantly active is platformio rebuilding the intellisense every 2-3 seconds ...

frankylynch, 2020-09-04:
Hi, I had v similar issues where is would build every 20 attempts .. I'm v new to this so no expert .. but ... two things [1] do you have the Arduino plugin installed for vs code? [2] the Grbl_Esp32.ino.cpp was being created and deleted every few seconds as you described... I disabled the auto rebuild intelligence .... after this its building and deploying consistently

lienbacher, 2020-09-09:
Dang @frankylynch that was it!! Disabling the "Auto Rebuild Autocomplete Index" setting in VSCode seems to have solved the issue! Just in case anyone runs into it: Click the settings button in the lower left corner, choose "Settings", type auto rebuild in the top bar and it should be the only setting left. Uncheck done. That was a hard one, I'm surprised so little other people run into this ...

closed 2020-09-09 by lienbacher

HuubBuis, 2020-09-09:
I have updated to the latest PIO version (5.0) and now I get the same error!

HuubBuis, 2020-09-09:
Updated the PIO ESP32 environment to 2.0.0. (i think) same error
Downloaded the last Grbl_ESP32 zip, same error. 
I will try again in a few days, maybe there is an PIO update. If not, I will install the previous PIO version!

lienbacher, 2020-09-09:
you commented on the wrong issue tough :)