bdring/Grbl_Esp32 #557: 4 axis external board

karoria, 2020-08-18:
Hi @bdring,
I've just become sponsor of this wonderful project. I was actually waiting for the board in attached image on tindie, but as I know, whole project is marching very fast on "6 pack" hardware alongwith I2S capability. Till the "6 pack" becomes mature and standard, can I get the board in this image? If not, at least its design files, so that I can make my own. For my 4 axis VMC, it suits perfectly till I add extra features to my machines.

[image]

bdring, 2020-08-18:
@karoria 

Thank you for the donation.

That board was designed to test RS485 and 0-10V spindle control. I like to load up test controllers with other new ideas to test them as well. Everything worked great. Most of the features and firmware are in use in other areas now.

I did not think the controller was versatile enough to justify an assembled order. I can post the source files and I have one I could send you for the cost of postage. Putting together a perfect BOM takes a lot of time so I post what I have.

The one I have is a v2.0. There was a footprint error on the isolated p/s on V1. V2 only has an RS485 spindle (no 0-10V or PWM)

karoria, 2020-08-18:
No problem. I will be ok with design files (schematic diagram) as I have some changes in my mind specific to my machine. I don't need RS485 spindle output. 0-10V is best.

bdring, 2020-08-18:
I uploaded the [source files here](https://github.com/bdring/4_Axis_External_Driver/blob/master/README.md).

You can see a tested [0-10V design here](https://github.com/bdring/6-Pack_CNC_Controller/wiki/0-10V-Output-Module)

karoria, 2020-08-18:
Thanks a lot. Rest I will manage.

closed 2020-08-18 by karoria