bdring/Grbl_Esp32 #573: Problems Compiling Firmware: polar_coaster

tinmahc, 2020-08-31:
A while ago I purchased a Grbl_ESP32 CNC Development Board (V3.1) from Tindie, knowing at some point I'd find a use for it.
I now have a need for a simple two joint scara machine, and I'm determined to use the development board and the awesome Grbl_ESP32 firmware.

I planing to build on the work the talented Mr Dring has done with the polar coaster and inverse kinematics. However, I'm not off to a good start. I'm getting the following Arduino IDE error message when I attempt to compile with polar_coaster.h as the defined machine in Machine.h 

`Alternatives for Custom/polar_coaster.cpp: []
ResolveLibrary(Custom/polar_coaster.cpp)o_build_771900\sketch\src\CustomCode.cpp:7:34: fatal error: Custom/polar_coaster.cpp: No such file or directory`

I'm able to compile (and upload) non polar_coaster configurations without issue, including test_drive.h and 3axis_v3.h
It seems that any configuration that needs a file from the Custom folder has the same compile error... Custom/....cpp: No such file or directory.

Hopefully I'm not missing something embarrassingly obvious.

I'm using an unmodified "main" branch that was downloaded today - only mod is to define machine in Machine.h
Arduino IDE is 1.8.13 (Windows 10)

MitchBradley, 2020-08-31:
We need to fix that, but in the meantime, perhaps you would consider compiling with PlatformIO, either under VSCode or from the command line. PlatformIO compiles are *much* faster than Arduino, so you will be happier in the long run, especially since you are likely to need many compilations to complete your coding project.

tinmahc, 2020-08-31:
Thanks for the suggestion, problem solved!
Have been meaning to test drive PlatformIO/VSCode for a while now, glad I've finally done it.

closed 2020-08-31 by tinmahc

Richaellima, 2022-04-23:
I have the same problem, how did you solve this error? Thanks...