bdring/Grbl_Esp32 #574: motor pins

sshahriyari, 2020-08-31:
Hi guys,
I can't find any data about wiring, especially for motor pins.

HuubBuis, 2020-08-31:
Look at the file: Grbl_Esp32/src/Machine.h, it uses files in the directory machines. 

I assume you need to connect de controller to a motor DRIVER

sshahriyari, 2020-09-01:
> Look at the file: Grbl_Esp32/src/Machine.h, it uses files in the directory Grbl_Esp32/src/machines.
> 
> I assume you need to connect the ESP32 controller to a motor DRIVER

many thanks HuubBuis!

I want to build a 6 axis machine. how can I configure the firmware and which pin is for Dir and Step of each axis?

HuubBuis, 2020-09-01:
There is a 6 axis configuration file  Grbl_Esp32/src/Machines/6_pack_stepstick_v1.h  that uses I2S stepstick drivers. You need a special board like this one [https://github.com/bdring/Grbl_Esp32/wiki/Hardware#mpcnc-controller-open-source](url) but there are others. Someone else must give you an advice on this.

You configure the firmware by setting the machine file in machine.h

The motors are connected like the most stepstick drivers using a 4 pin dupont connector.

sshahriyari, 2020-09-11:
> There is a 6 axis configuration file Grbl_Esp32/src/Machines/6_pack_stepstick_v1.h that uses I2S stepstick drivers. You need a special board like this one [...]

the link is dead.

I do not have access to the board and I want to design my custom board.

I just want to know which pin is for step and direction in the esp32 board (output pin)

joedirium, 2020-09-11:
You make take a look here: https://github.com/bdring/6-Pack_CNC_Controller