bdring/Grbl_Esp32 #580: Coolant overrides are not independent

MitchBradley, 2020-09-06:
Firmware: main as of 9/6/202

Repeatable problem: Coolant overrides do not control mist and flood independently if mist and flood are both on.

To duplicate the problem:
Use a machine config that has both COOLANT_MIST_PIN and COOLANT_FLOOD_PIN defined, with something attached to those pins so you can tell if they are on.

Try various combinations of M7, M8, and M9 on separate lines - that works correctly, with M7 turning on Mist, M8 turning on Flood, and M9 turning both off.

Now issue override commands (instructions below).  Starting with both off, issue the real-time command character 0xa0 to toggle the flood coolant.  The first one turns it off, the next turns it on, and so on.  Again starting with both off, issue 0xA1 to toggle the mist coolant.  It toggles as expected.

Now for the problem situation:  Toggle mist and flood until they are both on simultaneously.  Then try to toggle flood off with 0xA0.  It doesn't work; both stay on.  If you then immediately issue 0xA1, both go off at the same time. There are several confusing scenarios depending on how many times you try to toggle one when both are on.

The root cause of the problem is a deficiency in coolant_set_state(), which is capable of turning either on individually, but can only turn them off simultaneously.  That matches the semantics of M7, M8, M9, but is not sufficient for the override toggles where either can be turned off individually regardless of the other's state.

MitchBradley, 2020-09-06:
How to send real-time coolant override characters:

There may be GCode senders that support coolant overrides, but here is a way to do it from a terminal emulator.  Terminal emulators often have no obvious way to send "high-bit" characters like 0xA0, but there is a way to do it with TeraTerm Pro under Windows.  (There are probably others too; these instructions focus on TeraTerm Pro).

You have to edit the TERATERM.INI configuration file.  If there is a copy of that file in your Documents folder (/Users/<yourname>/Documents/TERATERM.INI), edit that file.  Otherwise copy /Program Files (x86)/teraterm/TERATERM.INI to /Users/<yourname>/Documents/TERATERM.INI and proceed with the edit there.

Look for the "MetaKey" variable and change it from "off" to "on".  Then look for Meta8Bit (should be right after MetaKey) and change that from "off" to "raw".

Then after you restart TeraTerm Pro you should be able to send 8-bit characters with the Alt key plus the ASCII character that has the desired 7 low bits.

For 0xA0 (toggle flood) that is 0x80 (high bit) + 0x20 which is "space", so to send 0xA0 you type Alt-<spacebar>

For 0xA1 (toggle mist) that is 0x80 (high bit) + 0x21 which is !, so to send 0xA1 you type Alt-! - and since ! is the shift of the 1 key, it is really Alt-Shift-1 .

MitchBradley, 2020-09-06:
There is a related problem where $G can report at most one of M7 M8 M9.  It should be able to say M7 M8 when both coolants are on, but it cannot.

bdring, 2020-09-07:
Seems to work for me on a real machine. 

All modes work
Toggling works
$G reporting works.

Thanks for the tips on Tera Term

MitchBradley, 2020-09-08:
Fixed by #581

commit 50ddc34d (2020-09-08): Fix #580 Coolant overrides (#581)

The approach was

a) Split up the CoolantMode enum, which had become complicated because
   it was trying to solve two incompatible requirements, into two
   enums GCodeCoolant (a pure enumeration for the use of the GCode
   parser) and CoolantState (a bitfield for runtime use where
   independent turn-off is possible via overrides)

b) Fixed coolant_set_state() so it can turn off coolant bits
   independently.  Previously it could turn them on independently,
   but only turn them off simultaneously.  That was fine for M7 M8 M9,
   but inadequate for realtime [... cut]
closed 2020-09-08 by MitchBradley