bdring/Grbl_Esp32 #582: RC Servos Hanging on Run

edplese, 2020-09-06:
What version of the firmware are you using?

main branch, fbf1732efae8c276d4b1d14c26ad6b0705339255, compiled with platform.io on a Mac.

Also have tested with a few older commits going back a couple weeks to see if it was a recent change.

Is the problem repeatable?

Yes

Under what conditions does the bug occur?

I have a 6 Pack board, version v1p1, with an RC servo module installed in module 3.  I started with the 6_pack_stepstick_v1.h config and added the servo configuration from the wiki and also removed the stepper configuration for axis A, B, and C: https://github.com/bdring/6-Pack_CNC_Controller/wiki/RC-Servo-BESC-CNC-I-O-Module.  All of the defaults were used for the GRBL configuration.

[6_pack_stepstick_v1.h.txt](https://github.com/bdring/Grbl_Esp32/files/5180422/6_pack_stepstick_v1.h.txt)

The stepper drivers work fine in this configuration.  When I issue a command to move the servo axes it seems to get stuck in a "Run" state which never ends:
```
g0a5
<Idle|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ>
ok
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ|Ov:100,100,100|A:S>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
```

The steppers work correctly.

The BESC on pin 1 of the module works correctly and when I connect a servo to it it moves when issuing "m3 s5000".  Viewing the pin on a scope I see the proper servo pulses.

The other pins 2, 3, and 4 for the A, B, and C axes all output a constant 3.3V and there are no pulses.  If I move axis A over to pin 1 it shows this same behavior.

I tried defining USE_SERVO_AXES as well but this no longer seems to be required and did not have any impact.

I am fairly new to Grbl_Esp32 so this may very well be a user error.  Any help would be appreciated!

bdring, 2020-09-06:
Paste your boot messages here

https://github.com/bdring/Grbl_Esp32/wiki/Requesting-Help#boot-messages

edplese, 2020-09-07:
Hi @bdring thank you for the link.  For entering the commands I am doing so through the WebUI.

Here are the boot messages:
```
[MSG:Grbl_ESP32 Ver 1.3a Date 20200828]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:6 Pack Controller V1 (StepStick)]
[MSG:Axis count 6]
[MSG:I2S Steps, Stream]
[MSG:Init Motors]
[MSG:X  Axis standard stepper motor Step:I2SO(2) Dir:I2SO(1) Disable:I2SO(0)]
[MSG:Y  Axis standard stepper motor Step:I2SO(5) Dir:I2SO(4) Disable:I2SO(7)]
[MSG:Z  Axis standard stepper motor Step:I2SO(10) Dir:I2SO(9) Disable:I2SO(8)]
[MSG:A  Axis RC Servo motor Output:4 Min:0.000mm Max:10.000mm]
[MSG:B  Axis RC Servo motor Output:16 Min:0.000mm Max:20.000mm]
[MSG:C  Axis RC Servo motor Output:27 Min:0.000mm Max:15.000mm]
[MSG:Using StepStick Mode]
[MSG:BESC spindle on Pin:GPIO(26) Min:0.90ms Max:2.20ms Freq:50Hz Res:16bits]

[MSG:Client Started]
[MSG:Connecting ednetx]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connected with 192.168.1.97]
[MSG:Connected with 192.168.1.97]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Probe on pin GPIO(34) Inverted:N]
```
And also the settings:
```
$Sta/SSID=ednetx
$Sta/Password=******
$Sta/IPMode=Static
$Sta/IP=192.168.1.97
$Sta/Gateway=192.168.1.1
$Sta/Netmask=255.255.255.0
$AP/SSID=GRBL_ESP
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=grblesp
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=STA
$Bluetooth/Name=btgrblesp
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Report/StallGuard=
$Spindle/Type=BESC
$Stepper/Pulse=4
$Stepper/IdleTime=250
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=Off
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=1.000
$GCode/MaxS=1000.000
$GCode/MinS=0.000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$X/StepsPerMm=50.000
$Y/StepsPerMm=100.000
$Z/StepsPerMm=100.000
$A/StepsPerMm=100.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=36000.000
$Y/MaxRate=1000.000
$Z/MaxRate=1000.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=2000.000
$Y/Acceleration=200.000
$Z/Acceleration=200.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/MaxTravel=300.000
$Y/MaxTravel=300.000
$Z/MaxTravel=300.000
$A/MaxTravel=100.000
$B/MaxTravel=100.000
$C/MaxTravel=100.000
$X/Current/Run=0.250
$Y/Current/Run=0.250
$Z/Current/Run=0.250
$A/Current/Run=0.250
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.125
$Y/Current/Hold=0.125
$Z/Current/Hold=0.125
$A/Current/Hold=0.125
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
```

bdring, 2020-09-07:
Manually send commands 

G0A0
G0 A10

Does the A servo move?

edplese, 2020-09-07:
Hi @bdring after sending the G0A0 (via WebUI) the servo does not move or hold position.  The signal pin is held high at 3.3v starting from poweron and this continues after the G0A0 command.  After the command is issued GRBL flips to the Run state and never exits until I reset the ESP32.
```
G0A0
ok
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ|WCO:0.000,0.000,0.000, nan, nan, nan>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ|Ov:100,100,100|A:S>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
<Run|MPos:0.000,0.000,0.000,0.000,0.000,0.000|FS:600,0|Pn:XYZ>
```

edplese, 2020-09-07:
I tried a simpler config on a different ESP32 and see the servo signals working correctly.  I am going to do a bit more testing on this to see if I can narrow down the issue a bit.

bdring, 2020-09-07:
Did you try G0A10?

What socket is it in?

edplese, 2020-09-07:
@bdring Thank you for your help with this today.  I figured out the issue.  I worked up from a simple config to the same full config header file on another ESP32 and it worked fine so I reset the EEPROM settings on the original broken one when I saw these messages:
```
$RST=* (probably only needed $RST=$)
[MSG:Servo calibration ($133) value error. 300.00 Reset to 100]
[MSG:Servo calibration ($134) value error. 300.00 Reset to 100]
[MSG:Servo calibration ($135) value error. 300.00 Reset to 100]
```
After restarting it the servos all worked perfectly.

As a bit more background on this I had originally loaded the stock `6_pack_stepstick_v1.h` firmware before modifying it for A, B, and C to be servo axes.  After this change I wasn't sure if a settings reset was needed and when I checked the settings all of the values looked correct like they had updated to be the servo default settings so I left it.

From the error it must have still had the 300.00 value in as those settings but when viewing the settings it showed as 100:
```
$A/MaxTravel=100.000
$B/MaxTravel=100.000
$C/MaxTravel=100.000
```
I'm good with closing this issue out now but wanted to pass along these details in case this isn't working as expected with regards to the settings.

Thank you again!

bdring, 2020-09-07:
The settings issue was due to original Grbl's rigid settings system and difficulty to add new settings. With the new settings system, we have more flexibility.  

There is a new branch called ServoSettings that has a new simpler way of working with servos. It should be merged with main in a week or two.

closed 2020-09-07 by bdring