bdring/Grbl_Esp32 #589: After updating to Pio Core 5.0 and solving previous compiling Problems I get "fatal error: HTTPClient.h: No such file or directory"

lienbacher, 2020-09-09:
After solving issue #554 and updating to the new pio core 5.0 I am running into an entirely new issue I cannot quite explain. 
compiling fails with this message:
```
Compiling .pio/build/release/lib1ac/ArduinoWebsockets/crypto.cpp.o
In file included from .pio/libdeps/release/ArduinoWebsockets/src/tiny_websockets/internals/ws_common.hpp:35:0,
                 from .pio/libdeps/release/ArduinoWebsockets/src/tiny_websockets/internals/wscrypto/crypto.hpp:3,
                 from .pio/libdeps/release/ArduinoWebsockets/src/crypto.cpp:1:
.pio/libdeps/release/ArduinoWebsockets/src/tiny_websockets/network/esp32/esp32_tcp.hpp:11:24: fatal error: HTTPClient.h: No such file or directory
```
The most interesting part though is if I open `esp32_tcp.hpp` and right-click "Go to definition" on `#include <HTTPClient.h>` it opens the file just fine in the arduino framework.

MitchBradley, 2020-09-10:
Have a look at this
https://github.com/bdring/Grbl_Esp32/pull/583

There is some confusion as to which libraries are the best to use.  The Grbl_Esp32 source tree includes copies of arduinoWebSockets (version 2.1.2) and ESP32SSDP (version 1.0) in the libraries/ folder at the top level.  But the platformio.ini lib_deps section includes the line arduinoWebSockets (now removed) which can cause platformio to install version 0.4.18.  If you remove that line, the local version will be used -but you might need to do a "clean" operation.

To make it more confusing, if you replace the line "arduinoWebSockets" with just "WebSockets", platformio will install the latest version 2.2.0 of the WebSockets library which also seems to work.  The naming is pretty confusing.  The platformio package named "arduinoWebSockets" comes from "https://github.com/gilmaimon/ArduinoWebSockets", while the "WebSockets" package comes from "https://github.com/Links2004/arduinoWebSockets".  One of them is "arduino..", the other is "Arduino..."  Isn't Open Source fun?

The other folder in libraries/ is ESP32SSDP/.  That too can be pulled in with a line in platformio.ini lib_deps - but the line is "ESP32SSPD" - note the transposition of the last two characters "PD" instead of "DP".  That comes from a typo in the library.properties file in the repo.  The upstream repo was recently - like yesterday - fixed to correct that error, but the platformio registry still has the old name and does not have the new one.

Anyway, if you delete both the lines "arduinoWebSockets" and "ESP32SSPD" from platformio.ini, it will probably work.

HuubBuis, 2020-09-10:
> Anyway, if you delete both the lines "arduinoWebSockets" and "ESP32SSPD" from platformio.ini, it will probably work.

This and doing a **_Clean_** (Project task) did solve the problem.

lienbacher, 2020-09-10:
Ahh thank you Mitch! Just switching to Devt branch also fixed the issue. Maybe a good idea to merge #583 into master? Just thinking since it fixes a breaking bug ...

closed 2022-03-02 by lienbacher